modelica/ModelicaStandardLibrary #4383: Invalid access of protected elements in SmoothStep example

maltelenz, 2024-04-11:
This:

https://github.com/modelica/ModelicaStandardLibrary/blob/a877f527e29c47e344d2c05e393b65c6c4eee01b/Modelica/Electrical/Analog/Examples/Lines/SmoothStep.mo#L92-L97

Accesses these protected elements:

https://github.com/modelica/ModelicaStandardLibrary/blob/a877f527e29c47e344d2c05e393b65c6c4eee01b/Modelica/Electrical/Analog/Lines/OLine.mo#L57-L65

and is hence illegal, as reported by System Modeler:

> Error: Modelica.Electrical.Analog.Examples.Lines.SmoothStep Invalid lookup of protected element. When looking up oLine50.L.i in scope Modelica.Electrical.Analog.Examples.Lines.SmoothStep, L is protected.
>   Protected elements can only appear as the first part of a component reference or path, and cannot be imported.

> Error: Modelica.Electrical.Analog.Examples.Lines.SmoothStep Invalid lookup of protected element. When looking up oLine50.C.v in scope Modelica.Electrical.Analog.Examples.Lines.SmoothStep, C is protected.
>   Protected elements can only appear as the first part of a component reference or path, and cannot be imported.

> Error: Modelica.Electrical.Analog.Examples.Lines.SmoothStep Invalid lookup of protected element. When looking up oLine5.L.i in scope Modelica.Electrical.Analog.Examples.Lines.SmoothStep, L is protected.
>   Protected elements can only appear as the first part of a component reference or path, and cannot be imported.

> Error: Modelica.Electrical.Analog.Examples.Lines.SmoothStep Invalid lookup of protected element. When looking up oLine5.C.v in scope Modelica.Electrical.Analog.Examples.Lines.SmoothStep, C is protected.
>   Protected elements can only appear as the first part of a component reference or path, and cannot be imported.

> Error: Modelica.Electrical.Analog.Examples.Lines.SmoothStep Invalid lookup of protected element. When looking up oLine1.L.i in scope Modelica.Electrical.Analog.Examples.Lines.SmoothStep, L is protected.
>   Protected elements can only appear as the first part of a component reference or path, and cannot be imported.

> Error: Modelica.Electrical.Analog.Examples.Lines.SmoothStep Invalid lookup of protected element. When looking up oLine1.C.v in scope Modelica.Electrical.Analog.Examples.Lines.SmoothStep, C is protected.
>   Protected elements can only appear as the first part of a component reference or path, and cannot be imported.

maltelenz, 2024-04-11:
Same thing here:

https://github.com/modelica/ModelicaStandardLibrary/blob/a877f527e29c47e344d2c05e393b65c6c4eee01b/Modelica/Electrical/Analog/Examples/Lines/CompareLineTrunks.mo#L91-L98

beutlich, 2024-04-11:
Broken by 839ab4de9c84a4a9bcbf070708f982969c84b484 for #4306. Not sure if the protected section was modified on purpose or accident. Given the commit message it does not look a necessary change.

casella, 2024-06-18:
I confirm also OpenModelica complains about that:
- https://libraries.openmodelica.org/branches/master/Modelica_trunk/files/Modelica_trunk_Modelica.Electrical.Analog.Examples.Lines.SmoothStep.err
- https://libraries.openmodelica.org/branches/master/Modelica_trunk/files/Modelica_trunk_Modelica.Electrical.Analog.Examples.Lines.CompareLineTrunks.err
- https://libraries.openmodelica.org/branches/master/Modelica_trunk/files/Modelica_trunk_Modelica.Electrical.Analog.Examples.Lines.SmoothStep.err

I'll make a PR to fix it.

casella, 2024-06-18:
Will be fixed by #4422 on master, then we need to back-port it to maint/4.1.x

casella, 2024-07-31:
Reopening until it is back-ported to maint/4.1.x

casella, 2024-08-16:
#4446 back-ported #4422, all set.

PR #4446 (2024-08-16): Fix regression introduced by #4306 by using public alias variables for initial equations in all examples using OLine
Backport of #4422 from master to main/4.1.x.

Resolves #4383
closed 2024-08-16 by casella