modelica/ModelicaStandardLibrary #4386: Wrong usage of quantities in Modelica.Electrical.Analog.Interfaces.IdealSemiconductor

AHaumer, 2024-04-16:
I suppose the following two lines
` 
  v = (s*unitCurrent)*(if off then 1 else Ron) + Vknee;
  i = (s*unitVoltage)*(if off then Goff else 1) + Goff*Vknee;
`
should be replaced by:
`
  v = s*(if off then unitVoltage else unitCurrent*Ron) + Vknee;
  i = s*(if off then unitVoltage*Goff else unitCurrent) + Goff*Vknee;
`
otherwise the quantities are horribly wrong?
It's the same for Modelica.Electrical.Analog.Interfaces.IdealSwitch.
Should we fix that for 4.1.0? 
What's your opinion, @christiankral @casella @dietmarw @HansOlsson ?

HansOlsson, 2024-04-16:
Currently I would say it is fine, and not an urgent bug - and it's not clear if it is a bug.

The s-parameter has unit "1" and we can uniquely determine that the number 1 has unit resistance in `v = (s*unitCurrent)*(if off then 1 else Ron) + Vknee;` One _could_ make it clearer by writing `v = (s*unitCurrent)*(if off then unitResistance else Ron) + Vknee;`, but it's clear if it is needed, and I'm more concerned with the unit-issues in the Spice-part.

dietmarw, 2024-04-16:
Definitely something for `4.1.1` or `4.2.0` since we are long past the code freeze for `4.1.0`.

AHaumer, 2024-04-16:
@HansOlsson 
but it's weird to fiddle around with "unitCurrent" and "unitVoltage" and leave it up to the tool to determine the unit of "1".