modelica/ModelicaStandardLibrary #4388: Two redundant arguments C and A in dormqr

tobolar, 2024-04-16:
In `Math.Matrices.LAPACK.dormqr`, there are used two arguments `C`and `A` being real matrices. Unfortunately, without description. 
Following the LAPACK'S DORMQR documentation and the routine's call, IMO there shall be just one argument (probably `C`) enabled.

HansOlsson, 2024-04-16:
Hm... https://netlib.org/lapack/explore-html/d7/d50/group__unmqr_ga768bd221f959be1b3d15bd177bb5c1b3.html seems to indicate that there are two arguments C and A.

Note that due to Modelica's stricter input/output semantics there are in two C-matrices in the Modelica function, C and Cout - but ideally that can be simplified in tools.

tobolar, 2024-04-16:
> Hm... https://netlib.org/lapack/explore-html/d7/d50/group__unmqr_ga768bd221f959be1b3d15bd177bb5c1b3.html seems to indicate that there are two arguments C and A.

Yes, actually a=A and c=Cout. C is only used to initialize Cout and some dimensions (m, n, ...)  If input/output semantics shall be the point, I would expect something like 
```
  input Real C[:, :];
  ...
protected
  Real A[size(C, 1), size(C, 2)] = C;
  ...
```
to overcome this.

beutlich, 2024-04-16:
A (together with tau as returned by dgeqrf) is an orthogoanl matrix to be multiplied with C. Seems all correct to me.

tobolar, 2024-04-17:
@beutlich True! I missed this by my mistake. Sorry for this.

closed 2024-04-17 by tobolar