modelica/ModelicaStandardLibrary #4389: Wrong input type in specificEnthalpy_pTX, temperature_phX, and density_phX

casella, 2024-04-16:
Consider this simple MWE:
```modelica
model TestFunctionDefault
  package Medium = Modelica.Media.Air.SimpleAir;
  Medium.SpecificEnthalpy h = Medium.specificEnthalpy_pT(1e5, 300);
end TestFunctionDefault;
```
This model compiles in Dymola 2024X and produces the expected result. However, when I try to compile it with OpenModelica, I get this error:
```
[1] 18:31:01 Translation Error
[Modelica.Media: 4706:7-4709:26]: Type mismatch for positional argument 3 in TestMedia.TestFunctionDefault.Medium.specificEnthalpy_pTX(X=fill(0.0, 0)). The argument has type:
  Integer[0]
expected type:
  Real[1]
```
This output looks a bit odd at first sight, what's wrong in calling specificEnthalpy_pT(1e5, 300)? I thought it was an OMC bug, but in fact it turns out to be an MSL issue.

specificEnthalpy_pT, as defined in [PartialPureSubstance](https://github.com/modelica/ModelicaStandardLibrary/blob/a877f527e29c47e344d2c05e393b65c6c4eee01b/Modelica/Media/package.mo#L4599), calls `specificEnthalpy_pTX(p, T, fill(0, 0))`, which is defined in [PartialMedium](https://github.com/modelica/ModelicaStandardLibrary/blob/a877f527e29c47e344d2c05e393b65c6c4eee01b/Modelica/Media/package.mo#L4446) as `specificEnthalpy(setState_pTX(p, T, X))`, so it calls `setState_pTX` with a third argument `X = fill(0, 0)`.

Unfortunately, setState_pTX for SimpleAir is defined in [PartialSimpleIdealGasMedium](https://github.com/modelica/ModelicaStandardLibrary/blob/a877f527e29c47e344d2c05e393b65c6c4eee01b/Modelica/Media/package.mo#L6085) as:
```modelica
  redeclare function specificEnthalpy_pTX
    "Return specific enthalpy from p, T, and X or Xi"
    extends Modelica.Icons.Function;
    input AbsolutePressure p "Pressure";
    input Temperature T "Temperature";
    input MassFraction X[nX] "Mass fractions";
    output SpecificEnthalpy h "Specific enthalpy at p, T, X";
  algorithm
    h := cp_const*(T - T0);
  end specificEnthalpy_pTX;
```
with input `MassFraction X[nX]` instead of `MassFraction X[:]`. Since by default `nX = size(substanceNames, 1) = 1`, it then expects an array of dimension one, but instead gets an array of dimension zero, coming from fill(0,0).

I'm not sure why Dymola does not complain about this issue, but for sure it needs to be fixed. This holds for functions `specificEnthalpy_pTX`, `temperature_phX`, and `density_phX` in both `PartialSimpleMedium` and `PartialSimpleIdealGasMedium`.

beutlich, 2024-04-16:
SimulationX also reports the dimension error

```
Error in TestFunctionDefault: Wrong argument dimension.
Help: Check.DimCallArg
 
h=Medium.specificEnthalpy_pT(100000.,300);
The argument 3 (fill(0,0)) in the function call has the wrong dimension ([0]-vector). Expected dimension [1]-vector.
```

I wonder why OMC deduced type `Integer[0]` from `fill(0,0)`. Should it be `fill(0.0,0)`?

perost, 2024-04-16:
> I wonder why OMC deduced type `Integer[0]` from `fill(0,0)`. Should it be `fill(0.0,0)`?

OMC would normally convert it to `fill(0.0, 0)` automatically, but that's done after checking if the types match. So the error message just uses the actual types rather than what the types would've been had they been compatible.

beutlich, 2024-04-19:
OK, there also is #4393 now.

casella, 2024-06-26:
This issue is causing regressons in OMC, so I would put it in MSL 4.1.0, I don't see any reason to postpone it to 4.2.0. The proposed modification is actually less restrictive, so there should be no backwards compatibility problems.

casella, 2024-06-27:
We need to cherry pick #4390 to maint/4.1.x so we get this fixed in the forthcoming release.

casella, 2024-07-31:
> We need to cherry pick #4390 to maint/4.1.x so we get this fixed in the forthcoming release.

@Esther-Devakirubai can you please take care of that? Thanks!

Esther-Devakirubai, 2024-08-01:
> > We need to cherry pick #4390 to maint/4.1.x so we get this fixed in the forthcoming release.
> 
> @Esther-Devakirubai can you please take care of that? Thanks!

@casella Yes Sure! Thanks

Esther-Devakirubai, 2024-08-16:
#4390 backported to maint/4.1.x by #4453

commit 9e9cd912 (2024-05-22): Fixes #4389
commit b94b49a7 (2024-08-16): Fixes #4389
closed 2024-08-16 by casella