modelica/ModelicaStandardLibrary #4424: Completely specify initial conditions for DynamicPipeEnergyConservationCheck and DynamicPipeEnergyConservationCheck2

casella, 2024-06-18:
The models 
```
ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck
ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck2
```
do not have fully specified initial conditions. These should be added to make sure that the initialization problem is not tool-dependent. This is probably also the reason why the first of the two model currently [fails](https://libraries.openmodelica.org/branches/master/ModelicaTest_trunk/files/ModelicaTest_trunk_ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck.err) in OpenModelica.

maltelenz, 2025-02-20:
For the record, WSM fails correctness checking for `ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck2` at initialization, but matches results afterwards.

casella, 2025-02-25:
EDIT: `ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck` actually doesn't have any state variables, because the default assumptions (which are used by all the pipes in the test) sets mass, momentum & energy balances as steady-state:

[image]

so there is actually no missing initial equation for that model. The reason why OMC fails is probably that it doesn't really like to switch off all the potential states with `stateSelect = StateSelect.prefer` as dummies. Very likely a tool issue.

Regarding `ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck2`, it indeed doesn't have fully specified initial states. In fact, these are the settings for the pipe dynamics options:

[image]

DynamicFreeInitial means that we have dynamic balances, but no intial equations. I can set it, e.g., to SteadyStateInitial, so that we get steady-state behaviour instead; note that the assert statment on the energy balance is carried out at the end of the transient, because it is only valid in static conditions. OpenModelica handles this without any problems. Unfortunately, Dymola aborts the code generation, I guess this is due to redundant initial conditions stemming from the state constraints due to the connection with fixed pressures at the outlet.

@HansOlsson can you please check that?

@maltelenz do you want to try what happens with WSL when you set system.energyDynamics = SteadyStateInitial?

maltelenz, 2025-02-25:
The two variables we had issues with:

[image]

[image]

and all tested variables using `SteadyStateInitial`:

[image]

@casella let me know if there is anything else you wanted.

casella, 2025-02-25:
Maybe we can just add a small pressure loss to the test case to avoid index problems. I'll take care of that.

casella, 2025-03-03:
@Esther-Devakirubai please back-port PR #4424 to maint/4.1.x

casella, 2025-03-07:
Backported by #4566, reference files updated.

PR #4547 (2025-02-27): Fixes for #4424
Added small pressure losses to avoid index reduction with fixed pressure boundaries. Changed to explicit steady state initialization to avoid unspecified initial conditions.

Fixes #4424
closed 2025-03-07 by casella