modelica/ModelicaStandardLibrary #4428: ModelicaTest.Media.TestOnly.WaterIF97_dewEnthalpy uses partial package

perost, 2024-06-20:
ModelicaTest.Media.TestOnly.WaterIF97_dewEnthalpy is defined as:
https://github.com/modelica/ModelicaStandardLibrary/blob/a9c6eabc42058a7b08c5a54f13ac0754ac610b4e/ModelicaTest/Media.mo#L953-L963
However, `Modelica.Media.Water.WaterIF97_fixedregion` is a partial package, so using it in a simultion model like this is not allowed. Presumably it should just be using one of the non-partial Water mediums instead.

casella, 2024-06-20:
This is not a big deal, as it just involves a test model and not a reusable class in the MSL. Nevertheless, the model is invalid, so it should be fixed.

casella, 2024-06-27:
Will be fixed by #4429, needs to be ported to maint/4.1.0 afterwards

beutlich, 2024-07-23:
Resolved by #4429 and backported by #4437.