modelica/ModelicaStandardLibrary #4431: Modelica.Magnetic.FluxTubes examples - comment from 4418

HansOlsson, 2024-06-28:
I agree with removing the signals and the new Interval for ArmatureStroke (in order to see the accelerations in the armature that happens in very short bursts).

But I don't see why we have so short Interval for the other two models - to me it seems 0.01 s would work as well. To me if someone sets a very small Interval I expect that there is something relevant on that time-scale, and I don't know which signal.

BTW: The documentation is broken for those two models (and likely have been for quite some time) - it refers to flange_b for components that only have flange.

_Originally posted by @HansOlsson in https://github.com/modelica/ModelicaStandardLibrary/pull/4418#pullrequestreview-2134384603_

casella, 2024-06-28:
@AHaumer can you please commen on that?

AHaumer, 2024-07-02:
I just compared two simulations and yes 100 ms is ok.
We just have to take care of the reference results ...
I'll have a look at the broken links.

AHaumer, 2025-02-11:
Broken links are cosmetics -> shifted to next release