modelica/ModelicaStandardLibrary #4451: Modelica.Blocks.Discrete.UnitDelay does not delay input

JayHuLBL, 2024-08-15:
The `Modelica.Blocks.Discrete.UnitDelay` does not delay the input as described in its documents. 

Given an example model as below, where the `unitDelay_MSL` is the delay block from MSL and the `unitDelay_Corrected` is the one with correction.
[image]
It has the results as:
[image]

The `unitDelay_MSL.y` is the same as the output from the sampling block, while the `unitDelay_Corrected .y` is the expected one.

casella, 2024-08-16:
@JayHuLBL where do we find the corrected model?

Could you please post the code that allows to replicate the issue? 

Thanks!

JayHuLBL, 2024-08-19:
@casella  Thanks for your response.

Please see the test model as below:
```
model UnitDelayAndSampleBlock "Example model for the UnitDelay block"
  Buildings.Controls.OBC.CDL.Reals.Sources.Ramp ramp1(
    duration=1,
    offset=0,
    height=6.2831852)
    "Block that generates ramp signal"
    annotation (Placement(transformation(extent={{-60,-10},{-40,10}})));
  Buildings.Controls.OBC.CDL.Reals.Sin sin1
    "Block that outputs the sine of the input"
    annotation (Placement(transformation(extent={{-20,-10},{0,10}})));

  Buildings.Controls.OBC.CDL.Discrete.Sampler sam(
    samplePeriod=0.05)
    "sampler of a continuous signal"
    annotation (Placement(transformation(extent={{40,-50},{60,-30}})));
  Modelica.Blocks.Discrete.UnitDelay unitDelay_MSL(
    samplePeriod=0.05)
    "Unit delay from MSL"
    annotation (Placement(transformation(extent={{40,40},{60,60}})));
  Buildings.Controls.OBC.CDL.Discrete.UnitDelay unitDelay_Corrected(
    samplePeriod=0.05,
    y_start=1)
    "Unit delay with correction"
    annotation (Placement(transformation(extent={{40,-10},{60,10}})));

equation 
  connect(ramp1.y,sin1.u)
    annotation (Line(points={{-38,0},{-22,0}}, color={0,0,127}));
  connect(sin1.y, unitDelay_Corrected.u)
    annotation (Line(points={{2,0},{38,0}}, color={0,0,127}));
  connect(sin1.y, sam.u)
    annotation (Line(points={{2,0},{20,0},{20,-40},{38,-40}}, color={0,0,127}));
  connect(sin1.y, unitDelay_MSL.u)
    annotation (Line(points={{2,0},{20,0},{20,50},{38,50}}, color={0,0,127}));

annotation (
    Icon(graphics={
        Ellipse(lineColor={75,138,73}, fillColor={255,255,255},
          fillPattern=FillPattern.Solid, extent={{-100,-100},{100,100}}),
        Polygon(lineColor={0,0,255}, fillColor={75,138,73}, pattern=LinePattern.None,
          fillPattern=FillPattern.Solid, points={{-36,60},{64,0},{-36,-60},{-36,60}})}),
     uses(Modelica(version="4.0.0")),
     experiment(StopTime=1, __Dymola_Algorithm="Dassl"));
end UnitDelayAndSampleBlock;
```

The corrected unit delay is from our development branch [issue3953_unitDelay](https://github.com/lbl-srg/modelica-buildings/blob/issue3953_unitDelay/Buildings/Controls/OBC/CDL/Discrete/UnitDelay.mo).

casella, 2026-11-10:
@JayHuLBL, as @HansOlsson wrote in #4651, changing UnitDelay as you propose is a non-backwards compatible change. We should not make such changes in MSL 4.x.y, which should be backwards compatible with MSL 4.0.0.

To me, the proper use of such blocks is that you first put a sampler, and then a unit delay. Currently, I'm more comfortable with #4650. Are you fine with that?

JayHuLBL, 2026-11-11:
@casella Thanks for looking into this issue again. I agree with you that the compatibility indeed would cause more trouble. The change in #4650 clarifies the assumption and it would be sufficient.