modelica/ModelicaStandardLibrary #4457: Initial values for polyphase inductor currents etc.

ceraolo, 2024-08-22:
Good modelica models have initial values defined, otherwise the results can be depending on the used tool and the assumptions it makes on those inital values.

When creating electric circuits, the most common state variables whose initial values are to be set are capacitor voltages and inductor currents. Modelica.Electrical.Analog.Basic capacitors allow setting initial voltages, and inductors intial currents.

Unfortunately the same does not occur with  Modelica.Electrical.Polyphase capacitors and inductors. 
Maybe one can set initial values for a polyphase inductor through modifiers, but this, if possible, is tricky. Personally, I could not find a way, at least in a three-wire circuit for which only two of the three currents must be set.

It seems to me that polyphase inductors and capacitors, and the other polyphase components having state variables, should have a way to set initial values from their masks, similar to their single phase counterparts.

casella, 2024-08-22:
@AHaumer, @christiankral what do you think?

AHaumer, 2024-08-25:
Lets investigate the polyphase inductor, the capacitor can be treated analogous.
The number of phases m can be choosen by the user. We have available the currents i[m] through the component. 
You have situations in which you are allowed to define all m initial currents (star-point connected), but if the star-point is not connected the sum of the m currents is zero and you are allowed to define only m-1 initial currents. 
How do you handle that? You can set only all m initial values at once. @ceraolo please give me a suggestion.
You have to consider not only the trivial case where the m or m-1 currents are initialized to be zero, 
maybe you want to initialize them with m or m-1 different initial values. 
Even if we try to motivate the tool to present an initialization menu:
Dymola 2024x doesn't do that, OM 1.23.1 does so.
But you can only initialize all m currents together (yes with different initial values) but either all of them fixed=true or all of them fixed=false. How do you handle the condition star-point not connected? 
@HansOlsson Do you have a suggestion from the language point-of-view?
IMHO the only way to handle that is to write initial equations:
`model Inductor
  extends Modelica.Electrical.Polyphase.Basic.Inductor(i(start=zeros(m)));
initial equation 
  //i[m] = zeros(m);
  i[1:m-1]=zeros(m-1);
end Inductor;`

AHaumer, 2024-08-26:
In MSL we try to keep the models as general as possible without restricting the freedom of usage.
I still doubt that we can cover all possible cases without restriction and without confusing the user:

- m wires, no neutral: We have to initialize m-1 currents
- m wires, separate neutral: We have to initialize m currents
- m+1 wires, the last one is the neutral: We have to initialize m currents

In the first case (m wires, no neutral) I wouldn't initialize one (may be the last) current to -sum(all other initial currents), this easily leads to numerical problems. We have to leave this up to the tool.

Specifying an array of initial currents i0[m-1] or i0[m] looks clumsy compared to the initialization menu presented automatically due to the alias variable. 

We should also take care of the following facts: 

- The user should be able to choose the index of the not initialized current.
- The user should be able to choose separately which currents are initialized fixed=true and which current fixed=false: There might be use cases where the user calculates the initial currents with some uncertainty.
- How to you obtain a fixed=false initialization?

I'm not sure whether we don't forget a special case which is not covered by the solution. 
What's about polyphase inductors used as subcomponents in transformers, machines, lines, ....
So why not leave it to the user to initialize the currents with initial equations? 
The user should know his system and therefore also know how to initialize the whole system.

ceraolo, 2024-08-28:
> In MSL we try to keep the models as general as possible without restricting the freedom of usage. I still doubt that we can cover all possible cases without restriction and without confusing the user:

I understand this. Two notes:
- the community of users has lived for many years without any possibility to initialize, i.e. the least "generality" possible. Only now an user (me) appeared which noted something lacking
- in the case of Polyphase library there still exist in MSL the option to resort to individual components from Analog.Basic.

So, any "restriction" in Polyphase is not restriction in general since we have individual components from Analog.Basic. Whenever Polyphase usage tends to become tricky, there is for the user this fallback option.

In reality, in the first version of my message I intended to propose a solution that allowed users to choose the wires to which to set initial values (the remaining to be computed by the tool). It was easy-peasy also from the GUI point of view.
Later I thought that it was overkill, for me.

This solution involved the user to input two arrays, one with the currents, the other with the indexes. the array would have m or m-1 elements: the tool will act correspondingly. 
It was still user's responsibility to decide whether to select m-sized arrays or m-1.

My position is that there must be a balance between generality and ease of use. That's why I finally resorted to propose that the balance current be always the m-th. 

But I'm not against the other way that I myself initially conceived.

AHaumer, 2024-10-13:
@casella you're right, we should go for your proposal 1. @christiankral what do you think?
Should I prepare a PR?

christiankral, 2024-10-14:
Yes, I agree to apply proposal 1. 

@AHaumer Please prepare a PR.

AHaumer, 2024-10-14:
I'll try. @ceraolo keep in mind:
We have to be sure that - for an important application; inductor with following Y-connection and floating starpoint - we are able to initialize m-1 currents with fixed=true and the last one with fixed_fals.

AHaumer, 2024-10-14:
Nice try ;-) but:
- OpenModelica 1.23.1 presents the initialization menu, but you choose the fixed-attribute for all currents together -> bad idea
- Dymola 2024x even doesn't present an initialization menu
Try it yourself. Sorry, as long as the tool developers don't do their homework I'm not willing to invest time. 
I'm pretty sure other tools have similar drawbacks. 
[PolyPhaseInductor.zip](https://github.com/user-attachments/files/17367692/PolyPhaseInductor.zip)

AHaumer, 2024-10-14:
Even another try with parameters for initialization:
- OpenModelica presents both the initialization parameters as well as the initialization menu - bad.
- Dymola 2024x presents the initialization parameters ok, but the fixed is really clumsy
How do other simulators behave? No I'm not going to invest any further time.
[PolyPhaseInductor.zip](https://github.com/user-attachments/files/17367862/PolyPhaseInductor.zip)

casella, 2024-10-14:
@AHaumer we shouldn't design the MSL based on the currently implemented GUIs, but rather the other way round: GUIs should properly support user-friendly editing of the MSL Modelica code. Bear in mind that this is for MSL 4.2.0, so there will be time to implement that feature.

casella, 2024-10-14:
@AHaumer I downloaded your first PolyPhaseInductor test package, but it doesn't seem to me to be fully implementing my proposal 1. I see the following code:
```modelica
model Inductor "Ideal linear electrical inductors"
  extends Modelica.Electrical.Polyphase.Interfaces.TwoPlug(i(start=zeros(m)));
...
```
but it should be
```modelica
model Inductor "Ideal linear electrical inductors"
  extends Modelica.Electrical.Polyphase.Interfaces.TwoPlug(i(start=zeros(m), fixed = fill(false, m)));
...
```
right?

HansOlsson, 2024-10-15:
An alternative would be:
`  Modelica.Units.SI.Current i2[m](start=zeros(m)) = i annotation(Dialog(showStartAttribute=true));`

(but hopefully with a better name). That should show up as a start-value. Or:

```
  Modelica.Units.SI.Current i2[m](start=zeros(m))  annotation(Dialog(showStartAttribute=true));
  extends Modelica.Electrical.Polyphase.Interfaces.TwoPlug(final i=i2);
```

AHaumer, 2024-10-15:
@casella your proposal won't make a difference since fixed=false is the default.
@HansOlsson I like your proposal, The only drawback:
The fixed attribute is presented as a checkbox and you are not able to propagate a Boolean variable or formulate an expression in the diagram window. We have to be able to set m-1 currents fixed=true and 1 false.
Could we drop that bothersome checkbox behavior?
@HansOlsson would you please discuss possible solutions in the language group?
@casella That would be the correct way to influence tool developers.
Otherwise I could use special language constructs and hope that tool developers adopt these.
[PolyPhaseInductor.zip](https://github.com/user-attachments/files/17375450/PolyPhaseInductor.zip)

casella, 2024-10-17:
> @casella your proposal won't make a difference since fixed=false is the default.

It would. What OMEdit does is to automatically generate a GUI item to change start and fixed attributes _if an explicit modifier exists_, unless of course it is final. So the presence of an explicit modifier is taken as a hint that the user may want to change it also via the GUI. I understand Dymola does the same, not sure about other tools.

The `showStartAttribute` annotation was introduced a long time ago in Modelica 3.2, but as far as I understand it was never introduced in the Modelica.Electrical, probably because the existing heuristic of showing the start attribute if there was an explicit start attribute modifier just worked and nobody bothered.

Note that a start attribute modifier can be introduced upon inheritance, e.g.
https://github.com/modelica/ModelicaStandardLibrary/blob/8e7876b5414475abab003c466ca49050812aeb01/Modelica/Electrical/Analog/Basic/Inductor.mo#L2-L3

while an annotation cannot, which would force us to define a new variable i2 and then add final i = i2, as suggested above by @HansOlsson. No big deal, except that one may wonder why the current that I need to fix is called i2, perhaps it is the square of the current? I find it a bit confusing.

My point is: the start attribute heuristic has been in place for inductors since the dark ages. Why not just using the same pattern for polyphase component? That would be the most straighforward solution to me. It just requires to add modifiers to the extends clause:

```modelica
model Inductor "Ideal linear electrical inductors"
  extends Modelica.Electrical.Polyphase.Interfaces.TwoPlug(i(start=zeros(m), fixed = fill(false, m)));
...
```
That is from a library point of view. Regarding tool support, I would say that tools should support both this construct and the showStartAttribute annotation also for arrays. Do we agree on that?

We could also decide to deprecate that practice and mandate the use of the showStartAttribute. To me, that won't be a particuarly good idea, because it would make a lot of existing library unusable (start attribute dialogs disappear), unless you add those attributes. It's ok to deprecate despicable practices, but this one doesn't seem despicable to me. To me, not bothering people with unnecessary non-backwards compatible changes of the language is a golden rule.

AHaumer, 2024-10-18:
@casella i2 is just a not very descriptive name for the alias current, in my example i named it iL.

I have no objections against both solutions, either the annotation or the pressure that all tools have to implement the behavior to present an initialization menu for an array variable if a vector of tart values is provided,

The only impediment is the fact that all tools have to **avoid presenting "fixed" as a checkbox,** otherwise you can't provide fixed=true for m-1 currents and fixed=false for the last one which is necessary in the case of start connection with star-point not connected i.e. sum of all m currents = 0.

casella, 2024-10-18:
> It is confusing if you can set the start values in the menu and the fixed attributes you have to set in text layer.

Sure. What I mean is, I wont' implement a convoluted solution using separate boolean parameter arrays such as as `fixed_start` in the library, but rather rely on the GUIs to allow modifying arrays of start and fixed attributes in a nice way.

There is time until the release of 4.2.0 for tool vendors to adapt to this.

@HansOlsson, @adea31, @eshmoylova, @maltelenz, @gerdkurzbach do you agree?

HansOlsson, 2024-10-18:
> > It is confusing if you can set the start values in the menu and the fixed attributes you have to set in text layer.
> 
> Sure. What I mean is, I wont' implement a convoluted solution using separate boolean parameter arrays such as as `fixed_start` in the library, but rather rely on the GUIs to allow modifying arrays of start and fixed attributes in a nice way.
> 
> There is time until the release of 4.2.0 for tool vendors to adapt to this.
> 
> @HansOlsson, @adea31, @eshmoylova, @maltelenz, @gkurzbach do you agree?

Yes. The model should be clear and the tools should adapt to that. Whether Dymola will do it for 4.2.0 release I cannot state at this time.

maltelenz, 2024-10-21:
I have a hard time following all the rounds in this issue. Do I understand it right that the proposal is to use the case from `inductor3` in the most recent test zip file?

```
  Modelica.Units.SI.Current iL[m](start=zeros(m), fixed = fill(false, m))=i "Inductor currents"
    annotation(Dialog(showStartAttribute=true));
```

For this case, System Modeler currently shows:

[image]

Where you get a nice array editor for the `start`:
[image]

And a slightly less nice expression editor for the `fixed`:

[image]

I'm not sure why we aren't showing the nicer array editor in this case, it seems like a tool issue.

If this is the proposed change, it seems fine.

AHaumer, 2024-10-21:
Yes @maltelenz you are right we tend to implement this version. 
The problem I see: At least Dymola and OM present a checkbox for the fixed attribute and it is cumbersome to propagate an array of Booleans (you have to switch to text mode, not very intuitive and confusing).
[image]
[image]
[image]
We have to be able to set the fixed attribute differently (not all array elements the same).

maltelenz, 2024-10-21:
Thanks @AHaumer for the summary.

It seems you have buy-in from Dymola (@HansOlsson), OM (@casella) and System Modeler (me, and also already reasonably working, even if not perfect) for the tools to adapt to such a change in the library.

christiankral, 2024-10-22:
> ```modelica
> model Inductor "Ideal linear electrical inductors"
>   extends Modelica.Electrical.Polyphase.Interfaces.TwoPlug(i(start=zeros(m), fixed = fill(false, m)));
> ...
> ```
>
> That is from a library [...]

This is my preference, as we avoid to add an additional redundant variable, which is not needed from a modeling point of view. The tools shall be able to handle such a case.

AHaumer, 2024-10-22:
We decided to give it a try: see #4488 for PolyPhase Capacitor and Inductor, tested in an example.
If this works well, the same style of initialization shall be implemented for PolyPhase.Basic.{Capacitor, Inductor, SaturatingInductor, MutualInductor, ZeroInductor, Transformer, VariableCapacitor, VariableInductor}, Modelica.Electrical.Machines.*, Modelica.Electrical.PowerConverters.*, Modelica.Magnetic.FundamentalWave.* but this seems to be a lot of work -> later.

AHaumer, 2024-10-24:
#4488 is merged, but let's keep this discussion open since the initialization shall get implemented in other models in the same style:
PolyPhase.Basic.{SaturatingInductor, MutualInductor, ZeroInductor, Transformer, VariableCapacitor, VariableInductor}, Modelica.Electrical.Machines., Modelica.Electrical.PowerConverters., Modelica.Magnetic.FundamentalWave.*

ceraolo, 2026-01-05:
Since #4488  is merged since October 2024, I 'd expected to see initial currents for Polyphase inductors to be already included in MSL 4.1, but this is not the case. 
I understand that there are other Polyphase components that need to be updated, but at least inductors and capacitors could have been made available, I reason.
May I ask when they are planned to be included?

ceraolo, 2026-01-07:
> > If initial currents are taken as structural, they cannot be changed without rebuilding the full model, which limits the usability of setting initial conditions.
> 
> That's just not true. Tested with [#4718]

It is true with my example Test2.mo (and Dymola 2026x)

ceraolo, 2026-01-07:
> see [#4718](https://github.com/modelica/ModelicaStandardLibrary/pull/4718) [@ceraolo](https://github.com/ceraolo) would you pls. have a look?

I think the example was very much needed, and it is perfectly conceived. I also like the explanations in the information box.

ceraolo, 2026-01-07:
> <img alt="Image" width="378" height="95" src="https://private-user-images.githubusercontent.com/6900273/532987593-ac1a10d3-36e7-4122-8819-dea3b52d8a24.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc4MTIzNjQsIm5iZiI6MTc2NzgxMjA2NCwicGF0aCI6Ii82OTAwMjczLzUzMjk4NzU5My1hYzFhMTBkMy0zNmU3LTQxMjItODgxOS1kZWEzYjUyZDhhMjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MDEwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjAxMDdUMTg1NDI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWY1NzZiOWQ5ZDY4NGZjMmQyYzdhNzJjZjhlZjRiN2UyZTIxOTlhNzY0ZTliNWFhMmJhMTQ2NjkxOWVjNjFkZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.8BCJjuFQESZBxA2wIFA0PEWv_GZKT-mnPi5_6cw7Mis">
> I suppose that's the cause that you can't change the initial values. 

I also suppose this.

When conceiving Test2 I on purpose tried to use only GUI: did not try to check the code, because I wanted to put myself in the position of a "basic" user of Modelica (like my students). At that point I still expected that the polyphase initialisation provided could be made without going into the text view.

Now that PolyphaseInitialisation is available, this basic user is warned that for three-wire systems they may need to do some coding, when you say:

`It depends on the tool whether you have to set the fixed attribute textually or graphically.
`

It seems to me that at the moment both OM and Dymola require going into the text view to set this attribute.

AHaumer, 2026-11-13:
merged #4718 with https://github.com/modelica/ModelicaStandardLibrary/commit/d3ebcae58233400564314f7d4d4513a0c8fd974f

closed 2026-11-13 by AHaumer