modelica/ModelicaStandardLibrary #4458: ClockedWithDiscretizedContinuousController uses clocked PI block with initial equation

henrikt-ma, 2024-09-03:
The `ClockedWithDiscretizedContinuousController` contains a clocked instance of `Modelica.Blocks.Continuous.PI`:
https://github.com/modelica/ModelicaStandardLibrary/blob/35ad28dd83fa36b5393fad54fa38c08e94ba6bb6/Modelica/Clocked/Examples/SimpleControlledDrive/ClockedWithDiscretizedContinuousController.mo#L30

However, this block contains an initial equation:
```
initial equation
  if initType == Init.SteadyState then
    der(x) = 0;
  elseif initType == Init.InitialState then
    x = x_start;
  elseif initType == Init.InitialOutput then
    y = y_start;
  end if;
```

This means that the clocked variable `y` violates one of the two very basic rules about synchronous variable initialization in https://specification.modelica.org/master/synchronous-language-elements.html#initialization-of-clocked-partitions:
> Variables in clocked partitions cannot be used in initial equation or initial algorithm sections.

I know that the specification is considered more or less flawed when it comes to achieving the goal of being able to "sample any continuous-time model",
> The goal is that every continuous-time Modelica model can be utilized in a sampled data control system. This is achieved by solving the continuous-time equations with a defined integration method between clock ticks. With this feature, it is for example possible to invert the nonlinear dynamic model of a plant, see Thümmel et al. ([2005](https://specification.modelica.org/master/bib.html#bib7)), and use it in a feedforward path of an advanced control system that is associated with a clock.

but while waiting for the specification to achieve this goal, the `ClockedWithDiscretizedContinuousController` should use a control block without initial equations.

casella, 2024-09-10:
See [MCP 0038](https://github.com/modelica/ModelicaSpecification/tree/MCP/0038/RationaleMCP/0038) regarding how to solve this issue in general

casella, 2024-09-17:
#4465 fixes the issue on master for MSL 4.2.0. @Esther-Devakirubai we should back-port it to maint/4.1.x so we also get this in 4.1.0.

casella, 2024-11-11:
After the discussion during the last MAP-Lang meeting, there was an agreement to update the specification and allow this kind of behaviour, see modelica/ModelicaSpecification#3577. See comment from @HansOlsson in [#4465](https://github.com/modelica/ModelicaStandardLibrary/pull/4465#issuecomment-2462581008).  Hence, no need to change the MSL source code.

closed 2024-11-11 by casella