modelica/ModelicaStandardLibrary #4464: Ramp Component Code Unexpected Behavior

mgithubk, 2024-09-06:
I tried using the ramp component and did not get ramp behavior. I compared the code in GItHub with the Modelica Ramp source code. Running them both showed that the JSML code provides a ramp up to height + offset but reverts back to offset once time > start_time + duration. The Modelica code does not.  For reference:
JSML Code:
function ramp(t,start_time,height,duration,offset)
	y = if start_time < t 
		 if t < start_time + duration
			 offset + (t - start_time) * height / duration
		 else
			 offset
		 end
		else 
		offset
		
		end
	return y
end

Modelica Code:

function modelicaRamp(t,start_time,height,duration,offset)
	y = offset + if t < start_time 
					0 
				 else 
					if t < start_time + duration 
						(t - start_time) * height / duration
					else 
						height
					end
				 end
end

beutlich, 2024-09-06:
I cannot follow why Modelica.Blocks.Sources.Ramp should behave _unexpectedly_. It's all documented and shown

[image]

You might use Modelica.Blocks.Sources.SawTooth with nPeriod set to 1 for "resettable" ramp behavior.

mgithubk, 2024-09-07:
Please excuse this comment, I put it in the wrong repo! The Modelica ramp is correct!

closed 2024-09-17 by casella