modelica/ModelicaStandardLibrary #4471: Modelica.Electrical.Analog.Examples.Resistor is not very simple

HansOlsson, 2024-10-01:
The documentation of Modelica.Electrical.Analog.Examples.Resistor says
> This is a very simple circuit consisting of a voltage source and a resistor.

I don't agree with that.
Clearly the model looks simple and the electrical part is in itself simple, but the thermal coupling makes the behavior more complicated:

If you look at the code generated by Dymola you see that there's a non-linear system of equations!

The reason is that the thermal part isn't connected to a heat capacitor as one would naïvely expect, but only to a linear thermal conductor. Since the resistor's resistance depend on temperature we therefore have a non-linear system of equations keeping the loss-power (that is `v*i` or `v^2/R(T)` right?) in balance with conducted heat power ( `G*(T-293.15)` ).

I see a number of possibilities:
- Just don't say that it is "very simple".
- Add a small heat-capacitor simplifying the model behavior, while keeping the thermal conductor (without the thermal conductor it would just keep heating until it burned).

The current model would be limit if we in the second possibility have a an epsilon-heat capacitor and let epsilon go to zero.

HansOlsson, 2024-10-01:
(oops - wrong labels, both started on 'd').

casella, 2024-10-01:
Well, the _electrical circuit_ is simple, I can't imagine anything simpler than a resistor connected to an ideal voltage source. The _circuit behaviour_ maybe is not completely trivial because of the dependency of the resistance on the quadratic losses, but the circuit and its behaviour are two different concepts. 

Surfing the web I even found "[A simple chaotic circuit with a light-emitting diode](https://www.researchgate.net/publication/309351711_A_simple_chaotic_circuit_with_a_light-emitting_diode/link/59664e4f458515e9af990cf8/download?_tp=eyJjb250ZXh0Ijp7ImZpcnN0UGFnZSI6InB1YmxpY2F0aW9uIiwicGFnZSI6InB1YmxpY2F0aW9uIn19)", so it seems that a _simple circuit_ can even show chaotic behaviour with strange attractors.

Maybe we should include it the Modelica.Electrical examples 😃

AHaumer, 2024-10-01:
@HansOlsson what's the problem with a nonlinear set of algebraic equations? 
@casella To be honest: The "simple chaotic circuit with a light-emitting diode" is _not simple_, it consists of 4 OpAmps besides the diode. Anyway, yes we could implement that circuit.

HansOlsson, 2024-10-02:
> @HansOlsson what's the problem with a nonlinear set of algebraic equations? 

There's no problem with the equations.
However, if a new user looks at the examples and sees a "very simple" circuit, and think that it will be a good starting point to understand models and then sees algebraic equations, they might be a bit surprised.

AHaumer, 2024-10-02:
@HansOlsson What's easier with an additional ODE?

IMHO the example _is simple_ and demonstrates:

- Ohm's law
- power dissipation of a resistor
- (linear) temperature dependency of the resistance

It can even be solved analytically - witt a linear temperature dependency of resistance it leads to a quadratic equation for temperature.
[ExamplesResistor.pdf](https://github.com/user-attachments/files/17227002/ExamplesResistor.pdf)

HansOlsson, 2024-10-02:
Ok, but even if "simple" could we drop the "very"?

casella, 2024-10-02:
No big deal, feel free to open a PR for that 😃