modelica/ModelicaStandardLibrary #4477: Chaotic behavior: Modelica.Electrical.Analog.Examples.ChuaCircuit

AHaumer, 2024-10-12:
I wonder whether the results of the above mentioned example can be compared between different simulators.
The default parameter settings lead to chaotic behavior. The results are extremely sensitive not only to parameters and initial conditions but also to tolerance and details of the solver. 
Choosing other parameter settings would either lead to periodic behavior or to a fix point.
The parameters are extremely unrealistic and can't be validated with a real circuit:
Who can get an inductance of 18 H and capacitances of 100 F and 10 F?
@casella mentioned another chaotic oscillator in another discussion (#4771).
Inspired by that, I implemented a few chaotic circuits with realistic parameters. 
They are even of didactic interest,
I wonder whether we should:
- Just newly parameterize the mentioned example to obtain periodic behavior, and mention in the documentation how to get chaos.
- Add my examples to Modelica.Electrical.Analog.Examples with parameters for periodic behavior and describe how to get chaos.
I'm pretty sure that examples with chaotic behavior are nice demos but a very bad base for comparing results between different tools.
@casella @HansOlsson @christiankral What's your opinion?

AHaumer, 2024-10-12:
You can test my models at [https://github.com/AHaumer/ChaoticCircuits](https://github.com/AHaumer/ChaoticCircuits)

AHaumer, 2024-10-13:
To demonstrate what I mean by "compared between different simulators" here we have a comparison between OpenModelica v1.23.1 and Dymola 2024x, both with tolerance = 1e-6:
[image]

casella, 2024-10-14:
@AHaumer, the Chua circuit is design on purpose to produce chaotic behaviour. There is nothing wrong with that, in fact, it was a fundamental discovery of 20th-century mathematics that you could get such kind of pseudo-random behaviour out of fully deterministic behavioural descriptions of systems dynamics. **Of course the default experiment annotation should allow to actually see that.**

And, BTW, it would be interesting not only to look at time plots of individual variables, but also to phase plots that allow to see the strange attractor that characterizes this system:
[image]
It is interesting to note that the time plot is highly sensitive to initial conditions and precision, but the shape of the strange attactor isn't.

I'm not worried at all by the sensitivity of the solution to the solver parameters etc., **because that's the whole point of this system**. The only thing we should be careful about is to avoid using such chaotic motion for regression testing and cross-tool checking. For that, we need to use a much shorter simulation interval, over which the trajectory instability does not have enough time to develop.

casella, 2024-10-14:
Regarding the values of the parameters, I checked the [reference](https://people.eecs.berkeley.edu/~chua/papers/Kennedy93.pdf) cited in the documentation. Indeed, the values of the parameters are:
- L = 18 mH (1/1000 of what we have)
- C1 = 10 nF (1/1e9 of what we have)
- C2 = 100 nF (1/1e9 of what we have)
- R0 = 12.5 Ohm (1000 of what we have)
- Ga = -0.757576 mS (1/1000 of what we have)
- I'm not really sure about Gb, since the values I see in the paper are apparently not directly correlated to the ones I see in the Modelica model

As a consequence, the simulation interval should be a few ms, rather than 50,000 s. 

I'm not sure why the circuit parameter were scaled that way, we should probably ask @christophclauss. Since we have a proper reference to a published paper, we should use the same parameters that are given there.

AHaumer, 2024-10-15:
I am rather sure about the parameters (maybe not 100% those from the paper but they work).
R.R=1900 leads to periodic behavior, R.R=1800 to chaotic behavior.
The interesting fact about that zoo of circuits is the dependence on parameters: 
Changing a certain parameter you may obtain a fixed point, periodic behavior or chaotic behavior. 
With a script you might even produce a bifurcation diagram.
For comparison the periodic behavior is much better suited, for demonstration change the parameter to see chaotic behavior.
For these parameters you get standard electronic components. Chua's diode you may implement with one or two operational amplifiers; see the solution in my repo [](https://github.com/AHaumer/ChaoticCircuits)
[ChuaCircuit.zip](https://github.com/user-attachments/files/17375718/ChuaCircuit.zip)

AHaumer, 2024-10-15:
See discussion in the language group [#3473](https://github.com/modelica/ModelicaSpecification/issues/3473) about verificationExperiment annotation.