modelica/ModelicaStandardLibrary #4489: Update matio for MSL 4.1.0

casella, 2024-10-25:
Following the discussion in #4480, I think we should update matio in MSL 4.1.0 to the latest version, to avoid shipping a library with known vulnerabilities. We haven't updated matio for vulnerabilities for almost four years, we shouldn't wait MSL 4.2.0 to do that.

This requires back-porting #4285 to maint/4.1.x (update to matio 1.5.26), as well as merging #4455 to master and backporting to maint/4.1.x if we want to update to 1.5.27; we should also re-build the matio binary, since we agreed to still ship the binaries with the MSL for 4.1.0.

@beutlich what do you think?

beutlich, 2024-10-25:
#4455 is the appropriate PR for master. This is just the sources, not the binaries.

casella, 2024-11-11:
@beutlich do you think you can provide the binaries for MSL 4.1.0? Otherwise, we'll keep whatever version of MatIO we had when you last built the binaries.

beutlich, 2024-11-11:
Can do, but which branch or which PR?.

casella, 2024-11-11:
In 4.1.0 we decided to still keep the compiled binaries in the git repo. As I understand this would require to
- merge #4497
- backport it to maint/4.1.x
- recompile the relevant binaries and commit them to the Resources directory

But maybe I missed the point of your question?

beutlich, 2025-01-02:
@casella I am still waiting for your review on #4506 to finally move on with this back-port issue.

casella, 2025-02-11:
@beutlich we approved #4506. Please proceed with back-porting of #4497 and #4506 to maint/4.1.x.

beutlich, 2025-02-11:
> [@beutlich](https://github.com/beutlich) we approved [#4506](https://github.com/modelica/ModelicaStandardLibrary/pull/4506). Please proceed with back-porting of [...]

Done by #4531. Also #4285 needed to be back-ported.

casella, 2025-02-24:
Should be done now.

closed 2025-02-24 by casella

casella, 2025-04-11:
Binaries updated by #4589.