modelica/ModelicaStandardLibrary #4494: replaceable data records

AHaumer, 2024-10-27:
I’m using replaceable data records to parametrize models: see enclosed Test-library.
It is desired to display the name of the used/redeclared data record or a string parameter of the used/redeclared data record on the icon of the model to help the user.
I tried 4 different annotations:
1.	Text(textString="%data.Type"),
2.	Text(textString="data.%Type"),
3.	Text(textString="data.%name"),
4.	Text(textString=DynamicSelect("Type", data.Type))
Dymola 2024x displays:
1.	Data0
2.	data.Type
3.	data.component
4.	Data1
OpenModelica v1.24.0 displays:
•	data.Type
•	data.Type
•	data.component
•	Type
I have choosen redeclare Test.DataRecords.Data1 data, therefore I would expect a display of “Data1”.
Only Dymola no. 4 is correct. I see now that no. 3 can’t be correct, it either displays the instance name of the model or the instance name of the data record. It seems that Dymola and OpenModelica interpret differently “editing state” and “non.editing state”.
18.6.6 “DynamicSelect has the syntax of a function call with two arguments, where the first argument specifies the value of the editing state and the second argument the value of the non-editing state. The first argument must be a literal expression. The second argument may contain references to variables to enable a dynamic behavior.“
I would be interested in the behavior of other tools (@maltelenz ), and what’s correct in terms of the specification (@HansOlsson ?).
@casella  what’s your opinion?
[Test.zip](https://github.com/user-attachments/files/17533563/Test.zip)

maltelenz, 2024-10-27:
Without having looked at tool behavior or your library yet (will do tomorrow), are you aware of the new (not really anymore) syntax  `%{data.Type}`?

maltelenz, 2024-10-28:
System Modeler displays:
* data.Type (we only match `%data`, which is a full record, so we don't render a value, falling back on the name of the "parameter")
* data.data.Type (`%Type` is a a parameter in `Component`, with the value `data.Type`. Seems correct.)
* data.component (clearly correct, tools agree)
* Type (we use the `editing state` here, the diagram is not dynamic)
* %data.Type (the specification is unclear, see below)
* data.Type (we show the value of `Type`, which `data.Type`. Seems correct, tools agree.)

It is currently unclear (to me) [in the specification](https://specification.modelica.org/maint/3.6/annotations.html#text) if `a.b` references are allowed, and if so, if they are allowed in the form `%a.b`, or only in `%{a.b}`. It seems OpenModelica also assumed only local parameters. It should be clarified what should be allowed.

For `DynamicSelect`, this has also come up in https://github.com/modelica/ModelicaSpecification/issues/3376 and mentioned in https://github.com/modelica/ModelicaSpecification/issues/3464 . My interpretation of the spec is that it is for animation (`Dynamic`!), so it shouldn't be used (or work) for a case like this.

AHaumer, 2024-10-28:
Thanks a lot @maltelenz !
So I have no solution for displaying a string parameter of the used/redeclared data record on the icon of the model,
and the spec is not clear enough. @HansOlsson how could we solve that?

maltelenz, 2024-10-28:
> And as far as I understand the `Text(textString="%{data.Type}")` seems like the good solution for this. If that is unclear we should clarify it.

Agreed! We should clarify this is OK in the spec.

I'm leaning towards saying `%data.Type` is not OK, similar to how `%'quoted'` is not OK.

casella, 2024-11-11:
@AHaumer sorry, no time for this until we finalize MSL 4.1.0. Will look at it ASAP.