modelica/ModelicaStandardLibrary #4512: Issue with ReferenceMoistAir Beyond 500°C

aci31, 2024-12-12:
Hi,

I recently faced an issue when using Modelica.Media.Air.ReferenceMoistAir.
When using the BaseProperties (so all volume based components, like the boundary or vessel), there is a "Index out of bound" issue happening only when the temperature is beyond 500°C.

I used Dymola 25x and MSL 4.0.0 (I have the same issue in Dymola 2020x with MSL 3.2.4).

Please find below a very simple example reproducing the error:
> model BugReferenceMoistAir
>   //The model just bug (Index out of bounds) when temperature reaches 500°C
>   package Medium=Modelica.Media.Air.ReferenceMoistAir;
>   Medium.BaseProperties medium;
> equation 
>   medium.p=1e5;
>   medium.T=273.15+time*1000;
>   medium.Xi=Medium.reference_X[1:Medium.nXi];
> 
>   annotation (uses(Modelica(version="4.0.0")));
> end BugReferenceMoistAir;

And the error message given by Dymola:
> Error: The following error was detected at time: 0.5
>   Index out of bounds
>   Failed condition: (j >= 1) && (j <= a.dims[0])
> The stack of functions is:
> Modelica.Media.Air.ReferenceMoistAir.Utilities.h_dis_pTX
> Modelica.Media.Air.ReferenceMoistAir.Utilities.h_pTX
> Modelica.Media.Air.ReferenceMoistAir.Utilities.u_pTX(100000.0, medium.T, medium.Xi)

The error triggered by **Modelica.Media.Air.ReferenceMoistAir.Utilities.h_dis_pTX()** comes from the fact that the functions (like Modelica.Media.Air.ReferenceMoistAir.Utilities.u_pTX()) in **BaseProperties** are called using Xi instead of X. Indeed, the function Modelica.Media.Air.ReferenceMoistAir.Utilities.h_dis_pTX() that is triggering the error is using X[1] and X[2] (obviously not available when using only Xi).
I duplicated the package Modelica.Media.Air.ReferenceMoistAir, and I modified all calls in BaseProperties to use X instead of Xi and it solved the error.

I looked in different Medium.BaseProperties, and all calls to functions using Xi... maybe we should correct that ?

casella, 2025-02-25:
@aci31 if you look into Modelica.Media.Interfaces.PartialMedium, you can see that all the _functions_ that accept a composition, e.g. `setState_pTX` or `specificEnthalpy_pTX` accept either `X` or `Xi` as input, e.g.:
https://github.com/modelica/ModelicaStandardLibrary/blob/7df356c9ce8b08da5858cf6037ce280dbae0d110/Modelica/Media/package.mo#L4129-L4136

the size of the composition is generic and is determined at compile time based on what argument you actually pass to it.

A posteriori, @hubertus65 recognized that it would have probably been a better design to only accept the full X vector (and maybe to move the whole `BaseProperties` machinery to `Modelica.Fluid`), but it is now too late to change the design without causing major backwards incompatibility issues, so we need to stick to that.

Now, the functions in `Modelica.Media.Air.ReferenceMoistAir` that implement this interface, e.g., setState_pTX:
https://github.com/modelica/ModelicaStandardLibrary/blob/7df356c9ce8b08da5858cf6037ce280dbae0d110/Modelica/Media/Air/ReferenceMoistAir.mo#L108-L121

are consistent with it. Most of the functions that are specific of this medium only, e.g.

https://github.com/modelica/ModelicaStandardLibrary/blob/7df356c9ce8b08da5858cf6037ce280dbae0d110/Modelica/Media/Air/ReferenceMoistAir.mo#L2877-L2935

are still consistent with that convention (i.e., they only use X[1]). The only exception is the "dis" functions, h_dis_pTX(), cp_dis_pTX(), s_dis_pTX(), h_dis_pTX_der(), which use X[1] and X[2].

For consistency, I think the best option is to replace X[2] with (1 - X[1]) in those functions.

@hubertus65, @arunkumar-narasimhan, do you agree with that?

casella, 2025-02-25:
Should be fixed by #4543

hubertus65, 2025-02-25:
Yes, Francesco, I agree. @casella

casella, 2025-02-25:
@Esther-Devakirubai please back-port to maint/4.1.x

casella, 2025-02-25:
@aci31 this is now fixed for MSL 4.1.0 and beyond. It will be hopefully released in a few weeks.

@beutlich thanks for bringing this issue to the foreground.

PR #4543 (2025-02-25): Fixed Utilities functions of ReferenceMoistAir to also work with Xi
Fixes #4512 by avoiding to use X[2] and X_der[2] in all functions that get X[:] as input (could be either X or Xi).
commit 9e8b7fb6 (2025-02-25): Fixed Utilities functions of ReferenceMoistAir to also work with Xi (#4543)

Fixes #4512
commit 8c14d6bd (2025-03-02): Fixed Utilities functions of ReferenceMoistAir to also work with Xi (#4543) (#4546)

Fixes #4512 for MSL 4.1.0
closed 2025-02-25 by casella