modelica/ModelicaStandardLibrary #4517: Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive is numerically fragile?

casella, 2025-01-09:
The model Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive seems to be prone to numerical issues.

- It works in Dymola 2024x with DASSL both at Tolerance = 1e-6 (as per the experiment annotation) and Tolerance = 1e-7 (stricter tolerance for the generation of reference trajectories)
- It works in Dymola 2025x with DASSL at Tolerance = 1e-6
- It fails in Dymola 2025x with DASSL at Tolerance = 1e-7 with error: "Next output point TOUT (=1.000000e+00) too close to initial/event time T (=1.000000e+00) to start integration."
- It works in Dymola 2025x at both 1e-6 and 1e-7 tolerance if other solvers are used (IDA/CVODE/Radau)
- It fails in OMC 1.25.0-dev-269 with DASSL and other solvers (IDA, ESDIRK) at Tolerance = 1e-6 and Tolerance = 1e-7 because of nonlinear solver failure on a strong component at time 0.1053

The failures in OMC can be attributed to the fact that the tool ends up with a large nonlinear torn system (116 unknowns, 23 iteration variables) that includes switching conditions, while Dymola is able to handle that more explicitly. However, also Dymola has issues with a normally robust solver (DASSL) if the tolerance is set to a too tight value.

Reducing the communication interval length doesn't seem to be beneficial.

casella, 2025-01-09:
@henrikt-ma, @maltelenz can you please report how WSL manages to run this model?

casella, 2025-01-09:
@HansOlsson do you have any clues about the failure in Dymola 2025x?

AHaumer, 2025-01-10:
It seems that we're discussing numerical issues of tools:

- Dymola up to 2024x has no problems
- Dymola 2025 brings up an error with tighter tolerance - @HansOlsson what's your opinion?
- OM seems to have problems
- Malte reports no problems with WSM

If this model is problematic also a lot of PowerElectronic models are problematic. 
I have no idea how to proceed if tool developers can't solve that.

HansOlsson, 2025-01-10:
> It seems that we're discussing numerical issues of tools:
> 
> * Dymola up to 2024x has no problems
> * Dymola 2025 brings up an error with tighter tolerance - @HansOlsson what's your opinion?

I wouldn't worry about the problems for that model with tight tolerance in Dymola 2025x for the MSL release.
We will investigate it later.

> * OM seems to have problems
> * Malte reports no problems with WSM
> 
> If this model is problematic also a lot of PowerElectronic models are problematic. I have no idea how to proceed if tool developers can't solve that.

henrikt-ma, 2025-01-13:
> It seems that we're discussing numerical issues of tools:
> …
> * Malte reports no problems with WSM

No, he said that reference results are matched *but* that there are warnings about non-converging event iterations.  This means that sketchy things are going on, and that the success of WSM cannot be taken as evidence of the model being good.

casella, 2025-03-06:
We should probably fix this locally for MSL 4.1.0, and then make a more thorough analysis of what could be better default values for Ron and Goff in Modelica.Electrical.Inverters for MSL 4.2.0. See [my comment in #4559](https://github.com/modelica/ModelicaStandardLibrary/issues/4559#issuecomment-2704012622).

casella, 2025-03-08:
PR #4572 addressed the problem by increasing the Ron and Goff parameters of the ideal diodes and thyristors.

@Esther-Devakirubai please back-port to maint/4.1.x

@MatthiasBSchaefer please update the reference file for Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive once the back-porting is done.

Esther-Devakirubai, 2025-03-10:
Backported by https://github.com/modelica/ModelicaStandardLibrary/pull/4579

casella, 2025-03-11:
All set

closed 2025-03-11 by casella