modelica/ModelicaStandardLibrary #4537: Wrong parameter in Magnetic/FluxTubes/Material

AHaumer, 2025-02-19:
The parameter sigma = **3**.22e6 in Magnetic/FluxTubes/Material/HysteresisTableData/M400_50A seems to be wrong, 
compared with sigma = **2**.22e6 in Magnetic/FluxTubes/Material/HysteresisEverettParameter/M400_50A.
sigma is the Electrical conductivity of material [S/m].
This value varies a lot with silicon doping. I'll try to search for parameters given by steel manufacturers.
@casella should we try to fox this for 4.1.0?

casella, 2025-02-24:
We could do it, the question is how many consequences does this have. Is it affecting any reference results? Do we need to report this in the release notes?

You can start by opening a PR on master, then we can decide whether to put it in 4.1.0 or not.

AHaumer, 2025-02-25:
I'll create a PR today's evening, also suggesting a short text for the release notes.
No this material M400_50A (there are 8 different material data sets in FluxTubes.Material.HysteresisTableData) is not used in an example or in Modelica.Test, therefore no reference results are affected,

casella, 2025-02-25:
@Esther-Devakirubai this will need to be back-ported to maint/4.1.x and a note added to the release notes about model fixes.

casella, 2025-03-06:
Back-port done by #4567

casella, 2025-04-11:
With 7dc176da1 this issue is mentioned in the critical fixes section.

closed 2025-04-11 by casella