modelica/ModelicaStandardLibrary #4548: Empty tank error in Modelica.Fluid.Vessels.BaseClasses.PartialLumpedVessel

MatthiasBSchaefer, 2025-02-26:
In
https://github.com/modelica/ModelicaStandardLibrary/commit/3434527b697b0ea8c59a08f7b7f935ec0e5f3b9e 
we changed the assert statement to fluidLevel > 0
but the default is defined:
input SI.Height fluidLevel = 0
So all examples, which do not modify this input fail (e.g. Modelica.Fluid.Examples.TraceSubstances.RoomCO2 ,see https://www.ltx.de/download/MA/Compare_MSL_v4.1.0/Testruns/Dymola/Modelica/_report/Modelica.Fluid.Examples.TraceSubstances.RoomCO2/testcase_report.html#)

MatthiasBSchaefer, 2025-02-26:
Especially the example Modelica.Fluid.Examples.Tanks.EmptyTanks which should 
"Show the treatment of empty tanks" (=description string of the example) probably won't work without a fluidLevel of (exactly) zero.

HansOlsson, 2025-02-26:
The simplest would be to change the assert to `>= 0`.

MatthiasBSchaefer, 2025-02-26:
> The simplest would be to change the assert to `>= 0`.

I tested it and it would solve all failures exept for Modelica.Fluid.Examples.Tanks.EmptyTanks.

But i am not sure if exactly zero means a singularity in the energy balance equations again

casella, 2025-02-27:
I'm afraid I misunderstood the scope of the partial classes involved in the making of the OpenTank model. The `fluidLevel` variable is defined in PartialLumpedVessel

https://github.com/modelica/ModelicaStandardLibrary/blob/b4a4b0336f8759418e916f29491f14789d9db905/Modelica/Fluid/Vessels.mo#L269-L270

and is meant to be used to deal correctly with ports at different heights that may be shut off by the automatic check v(alve behaviour if the level gets below their heigth. However, this base class is also used for models such as ClosedVolume (used in the TraceSubstances.RoomCO2) where the "level" is not used, so it should have a default.

So, the assertion should not be put on PartialLumpedVessel.fluidLevel, which is too generic. It should rather be put in the OpenTank model, where `level` (which is then passed as value for fluidLevel when extending from the base class) has a definite meaning:
- a cylindrical shape with vertical axis is assumed
- with a flat bottom
- hence volume = level * crossArea

BTW, this is not the only possible choice of semantics of level. Another tank model could not have such a cylindrical shape, and not necessarily level = 0 implies volume = 0. This is just a feature of OpenTank. Hence, level > 0 should be asserted in OpenTank. Once again, it needs to be > 0, not >= 0, because the OpenTank model is ill-defined at zero level.

Additionally, the Modelica.Fluid.Examples.Tanks.EmptyTanks should be modified in the same way the ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank was modified, i.e. by having a small but positive height for the bottom port that prevents the tank from getting completely empty.

I will follow up with a PR that fixes all these issue.

casella, 2025-02-27:
Thanks @MatthiasBSchaefer for pointing this out!

casella, 2025-03-01:
PR #4557 should fix this. Once that PR is merged in, we need to 
- [ ] re-generate the reference results and add them to the 4.1.0 branch
- [ ] back-port #4557 to maint/4.1.x

casella, 2025-03-05:
@Esther-Devakirubai please back-port this to maint/v4.1.x

@MatthiasBSchaefer, @GallLeo please re-generated reference results once that is done and update the 4.1.0 branch accordingly.

HansOlsson, 2025-03-06:
Just a notice that we tested the current 4.1.0-branch with Dymola and found that the current state for this problematic.

In particular the new assert caused problems for:
* Modelica.Fluid.Examples.TraceSubstances.RoomCO2
* Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls
* FMI-simulation for Modelica.Fluid.Examples.Tanks.EmptyTanks
* Other libraries e.g., using Modelica.Fluid.Machines.SweptVolume, inheriting from PartialLumpedVessel and starting with level 0. I haven't checked if there are problematic cases using other models.

I assume most of these will be fixed by the PR, but especially the FMI-case and other libraries requires more testing afterwards.

casella, 2025-03-06:
@HansOlsson this is expected, because the min attribute was put in the wrong base class, and that has bad side effects on other models. #4557 fixes that, once it is back-ported, these issues should be resolved.

casella, 2025-03-06:
Should be OK now.

casella, 2025-03-07:
Back-ported in #4568, all set.

commit 2c674e0b (2025-03-05): Fixes #4548 (#4557)

* Remove assert on fluidLevel in PartialLumpedVessel, that does not define a minimum level (nor it should!)

* Add the assertion on positive level in OpenTank

* Moved height of fluid port of tank1 0.001 m above the tank bottom to avoid emptying it completely

* Tank2 starts at a small positive value, made it more realistic (1 mm) rather than 0.1 nm, which doesn't make much sense

* Added some documentation to the example
closed 2025-03-07 by casella