modelica/ModelicaStandardLibrary #4552: Bug in dp_volumeFlowRate_MFLOW() function

casella, 2025-02-26:
While trying to solve #4424, I discovered that the pressure/flow behaviour of the Modelica.Fluid.Fittings.GenericResistances.VolumeFlowRate model when I set a = 0 and b > 0 was not 

dp = b*V_flow

as claimed by the documentation, but rather dp = 1/b*V_flow, which of course caused the pressure losses to be wrong by many orders of magnitude. The issue can be traced back to this chunk of code:

https://github.com/modelica/ModelicaStandardLibrary/blob/9e8b7fb633a145e3f2a82f2a1d887cb3d7da72ba/Modelica/Fluid/Dissipation.mo#L4618-L4625

where the last line should be dp/b rather than b*dp.

Probably nobody has ever used this component in this way, i.e. to obtain a linear pressure-flow characteristic, otherwise we would have heard about that. This needs of course to be fixed in MSL 4.1.0.

casella, 2025-02-26:
@Esther-Devakirubai, once #4533 is merged in, it will need to be back-ported to maint/4.1.0@

Esther-Devakirubai, 2025-03-10:
Backported by https://github.com/modelica/ModelicaStandardLibrary/pull/4580

casella, 2025-04-11:
43f410176 now mentions it in the Release Notes.

PR #4553 (2025-03-07): Fixed wrong expression in dp_volumeFlowRate_MFLOW
Fixes #4552.
closed 2025-04-11 by casella