modelica/ModelicaStandardLibrary #4554: Asserts on Real values with test models are too tight

casella, 2025-02-26:
There are two tests in ModelicaTest.Fluid.TestComponents that have too tight asserts. One is Machines.TestLinearPower:

https://github.com/modelica/ModelicaStandardLibrary/blob/b4a4b0336f8759418e916f29491f14789d9db905/ModelicaTest/Fluid/TestComponents/Machines/TestLinearPower.mo#L7

which is a numerical hazard, because `p` is computed by a linear equation, so the actual computed value could be something like 23.000000000001 or 29.999999999999 or something like that, which would of course be OK but would trigger the assertion failure.

The other one is ModelicaTest.Fluid.TestComponents.Pipes.DynamicPipeEnergyConservationCheck2:

https://github.com/modelica/ModelicaStandardLibrary/blob/b4a4b0336f8759418e916f29491f14789d9db905/ModelicaTest/Fluid/TestComponents/Pipes/DynamicPipeEnergyConservationCheck2.mo#L128-L131

where it is not clear whether the second part of the assertions will ever be checked, since 
https://github.com/modelica/ModelicaStandardLibrary/blob/b4a4b0336f8759418e916f29491f14789d9db905/ModelicaTest/Fluid/TestComponents/Pipes/DynamicPipeEnergyConservationCheck2.mo#L148

mestinso, 2025-03-08:
Quick comment on the second one (since I built that model): at least in Dymola that type of approach will trigger an error at exactly 500 seconds. Not sure what the modelica spec has to say about this, but we could increase the experiment time to 1000 seconds if there is concerns about how other tools handle this.

casella, 2025-03-08:
We had some discussion during the last MAP-Lib meeting and the agreement was to use `when terminal()` to make sure the asserts are triggered. I opened PR #4555 to do that, @beutlich raised some concerns, I'll update the PR based on that.

> at least in Dymola that approach will trigger an error at exactly 500 seconds (if the right side of the or statement is false). I'm not sure what the Modelica spec has to say about this

I guess nothing specifically about this scenario. My general understanding is that relying on two reals being exactly equal to each other to take some action, such as actually doing the check on the energy balance, is not appropriate. The only exception is checking for parameters that have a binding equation to be non-zero: somthing like assert(p > 0, "Parameter needs to be positive") is guaranteed to be triggered if p is zero, because Real p = 0; does not involve any possible rounding errors, but actually sets all bits of the IEEE 745 representation to zero, so that p > 0 fails.

I would prefer to steer clear from such situations, at least in the MSL.

casella, 2025-03-08:
PR #4555 updated, waiting for feedback from @beutlich.

@Esther-Devakirubai, once merged in, this should be back-ported to maint/4.1.x

casella, 2025-03-09:
#4555 merged in, waiting for back-port to maint/4.1.x.

Esther-Devakirubai, 2025-03-10:
> PR [#4555](https://github.com/modelica/ModelicaStandardLibrary/pull/4555) updated, waiting for feedback from [@beutlich](https://github.com/beutlich).
> 
> [@Esther-Devakirubai](https://github.com/Esther-Devakirubai), once merged in, this should be back-ported to maint/4.1.x

Backported by https://github.com/modelica/ModelicaStandardLibrary/pull/4581

casella, 2025-03-10:
All set.

commit cf4c2e33 (2025-03-09): Fixes #4554 (#4555)

* Addresses the first part of #4554

* Addresses second part of #4554
commit 78b6f5eb (2025-03-10): Fixes #4554 (#4555) (#4581)

* Addresses the first part of #4554

* Addresses second part of #4554
closed 2025-03-10 by casella