modelica/ModelicaStandardLibrary #4558: Shorten reference results for Chua circuit model

casella, 2025-03-01:
OpenModelica fails the verification against the reference files for Modelica.Electrical.Analog.Examples.ChuaCircuit, see e.g. [c1.v](https://libraries.openmodelica.org/branches/master/Modelica_4.1.0-beta.om/files/Modelica_4.1.0-beta.om_Modelica.Electrical.Analog.Examples.ChuaCircuit.diff.C1.v.html). 

[image]

The reason is that Chua circuit is specifically built to show chaotic behaviour, and in fact the OMC computed trajectory diverges sharply from the reference trajectory 

This is not new, it was already mentioned years ago in #1741, #3477, #4477, as well as in MAP-LIB_ReferenceResults#4. Unfortunately no action was taken so far, despite widespread consensus that a shortened reference file should be supplied.

In #4341 we already decided to generate a shortened reference file for BevelGear1D, this is already done in the current 4.1.0 references. We should also do it for Modelica.Electrical.Analog.Examples.ChuaCircuit. Based on the results observed above, I would suggest shortening the time to 4000 s.

@MatthiasBSchaefer, @GallLeo can you please take care of that?

Thanks!

beutlich, 2025-03-03:
This seems like a duplicate of https://github.com/modelica/MAP-LIB_ReferenceResults/issues/4.

AHaumer, 2025-03-03:
@casella for my understanding:
Do we 1. just shorten the reference result, or do we 2. set a shorter stop time in the example? 
ad 1. Do we have problems if the length of the reference result doesn't match the stop time?
ad 2. Then I see the problem that we need a "normal" stop time for demonstrating chaotic behaviour and a second stop time for regression testing.

maltelenz, 2025-03-03:
@AHaumer  We just shorten the reference result, like @GallLeo did [here](https://github.com/modelica/ModelicaStandardLibrary/issues/4341#issuecomment-2425693263).

MatthiasBSchaefer, 2025-03-04:
see https://github.com/modelica/MAP-LIB_ReferenceResults/pull/19

casella, 2025-03-05:
> This seems like a duplicate of [modelica/MAP-LIB_ReferenceResults#4](https://github.com/modelica/MAP-LIB_ReferenceResults/issues/4).

Indeed, but that was a general discussion, whereas here I just wanted to point out a specific issue that must be solved now.

casella, 2025-03-05:
> [@casella](https://github.com/casella) for my understanding: Do we 
> 1. just shorten the reference result

Yes, as we did already for BevelGear1D
  
> or do we 2. set a shorter stop time in the example?

No, the example is meant to actually show chaotic behaviour. There's nothing wrong with that. The only thing we should avoid is trying to do regression testing on it.

> ad.1. Do we have problems if the length of the reference result doesn't match the stop time?

Hopefully not, CSV-compared has been fixed specifically to handle that.
 
> ad.2 Then I see the problem that we need a "normal" stop time for demonstrating chaotic behaviour and a second stop time for regression testing.

Well, to me it's not a problem, but rather a requirement. Which could also hold for other kind of models, e.g. high-frequency oscillators. See modelica/ModelicaSpecification#3473 for a (long, unresolved) conversation about how to address it.

casella, 2025-03-05:
Done in modelica/MAP-LIB_ReferenceResults#19.

closed 2025-03-05 by casella