modelica/ModelicaStandardLibrary #4559: Regression: SMPM_Braking in MSL 4.1.0 fails due to chattering in OpenModelica

casella, 2025-03-01:
Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking in MSL 4.0.0 [simulated successfully](https://libraries.openmodelica.org/branches/master/Modelica_4.0.0/files/Modelica_4.0.0_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking.sim) in OpenModelica, with results matching the reference files.

It [fails instead](https://libraries.openmodelica.org/branches/master/Modelica_4.1.0-beta.om/files/Modelica_4.1.0-beta.om_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking.sim) in MSL 4.1.0 with the following error message:

Chattering detected around time 0.523614405406..0.523625675045 (100 state events in a row with a total time delta less than the step size 0.0001). This can be a performance bottleneck.
The zero-crossing was: diodeBridge2mPulse.diode_n.idealDiode[2].s < 0.0

This may be a tool issue, but was apparently triggered by some change in the library at some point. I will do some bisection to figure out when it happened.

@maltelenz does it work for you in WSM?

maltelenz, 2025-03-03:
This example does not work consistently in WSM. We have always had (platform dependent) issues with this model over time.

For the 4.1.0 version, we have issues on `Windows-64` (correctness) and `Linux-64` (looks like it could be chattering). Currently, it does work on the platforms `macOS-Intel`, `macOS-ARM`, `Windows-32`, `Linux-32`.

AHaumer, 2025-03-03:
I tested this example (which was implemented several years ago) both from MSL 4.0.0 and 4.1.0, both with OM 1.24.3 64 bit and Dymola 2025x 64 bit without any problem (ok OM reports "Homotopy solver Newton iteration: Maximum number of iterations reached at time 0.000000, but no root found.", but the simulation runs correctly,)
@casella I don't understand what's the root for these problems. @christiankral do you have any idea?

casella, 2025-03-03:
Interesting.

Indeed, I can also run Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking from MSL 4.1.0-dev successfully with OMC on my Windows 11 PC, while it [fails due to chattering](https://libraries.openmodelica.org/branches/master/Modelica_4.1.0-beta.om/files/Modelica_4.1.0-beta.om_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking.sim) on the CI running under Linux.

This is consistent with @maltelenz that reports platform-dependent problems on WSM.

casella, 2025-03-04:
Using OMC, I checked the diodeBridge2mPulse.diode_n.idealDiode[2].s variable around time = 0.523614405406...0.523625675045, which is where the MSL 4.1.0 model undergoes 100 switching on Linux, on my Windows PC. This is the event log:

[image]

This is a first plot in the 0.52...0.59 time interval

[image]

Zoom in 

[image]

Zoom in

[image]

Zoom in

[image]

So, it seems that this variable is "bouncing" three times at increasing frequency around s = 0, until it finally breaks through. The last bounce has a really small amplitude (5e-6), so maybe it's really borderline in terms of event detection, and possibly depending on OS-specific numerics.

I then tried to switch off the equidistant time grid output, so I can see the actual DASSL integration steps. If I do so, the simulation runs until time = 0.7227, then I get a ton of events, the last two at times: 0.726475568508 and 0.726475568896 (which are really, really close), and then the simulation crashes.

This is the overall behaviour of that variable

[image]

and this is a zoom of the last 0.15 s:

[image]

@AHaumer it seems pretty clear to me that the final state of this simulation is a singular state where all voltages and currents on the variable resistor get to zero, which is also the singular point of the ideal diodes. This seems pretty stretched to me from a numerical point of view.

I guess we should make this a bit more numerically robust by increasing RonDiode and/or decreasing GoffDiode. For example, if I set them both to 1e-4 instead of the default 1e-5, the simulation runs smoothly until the end also with no equidistant time grid.

casella, 2025-03-04:
See PR #4562. Except for the increased robustness, the results are virtually identical, here I compare the current in the variable resistors in the two cases and it's very close

[image]

[image]

If we accept #4562, we should probably re-generate the reference results.

@maltelenz please check if the changes in #4562 makes the model run OK in WSM.

casella, 2025-03-05:
@AHaumer if you are fine with PR #4562 I think you and @maltelenz can approve it, so we can move forward towards the release. Thanks!

maltelenz, 2025-03-05:
Reopening since we need a backport to 4.1.0. @Esther-Devakirubai can you create a PR?

AHaumer, 2025-03-05:
I suppose we should apply the same patch to

maltelenz, 2025-03-05:
@AHaumer In WSM we have the same historical issues with that model, so I'm in favor of applying the same approach if it is possible.

AHaumer, 2025-03-05:
@christiankral @casella @maltelenz I suppose w should apply the same patch to cure #4517 - what's your opinion?

casella, 2025-03-08:
Summary:
- PR #4562 fixed the problem for Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking. Back-ported to maint/4.1.x by #4570 
- PR #4563 fixed potential issues in the analogous Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_Braking

@Esther-Devakirubai please back-port #4563 to maint/4.1.x

@MatthiasBSchaefer please re-generate reference files for both models in MSL 4.1.0

casella, 2025-03-08:
> Therefore it's questionable to change the default parameters which would change the results of user models.

Agreed. That would have been a bad idea, so close to the release date.

> BTW what does "numerically fragile" mean? 

Good question! I opened a discussion ticket on that, see #4577

> I'm pretty sure some enhancements in the solver of a tool also solves the problem.

Not necessarily. I think we should not have blind faith in tools with numerical superpowers to solve badly conditioned problems, see #4577.

MatthiasBSchaefer, 2025-03-10:
> Summary:
> 
>     * PR [Attempt to fix #4559 by increasing RonDiode and GoffDiode #4562](https://github.com/modelica/ModelicaStandardLibrary/pull/4562) fixed the problem for Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking. Back-ported to maint/4.1.x by [Attempt to fix #4559 by increasing RonDiode and GoffDiode (#4562) #4570](https://github.com/modelica/ModelicaStandardLibrary/pull/4570)
> 
>     * PR [Same patch as #4559 to Electrical.Machines #4563](https://github.com/modelica/ModelicaStandardLibrary/pull/4563) fixed potential issues in the analogous Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_Braking
> 
> 
> [@Esther-Devakirubai](https://github.com/Esther-Devakirubai) please back-port [#4563](https://github.com/modelica/ModelicaStandardLibrary/pull/4563) to maint/4.1.x
> 
> [@MatthiasBSchaefer](https://github.com/MatthiasBSchaefer) please re-generate reference files for both models in MSL 4.1.0

New references for Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking according to [Attempt to fix #4559 by increasing RonDiode and GoffDiode (#4562) #4570](https://github.com/modelica/ModelicaStandardLibrary/pull/4570) is already merged in [#20](https://github.com/modelica/MAP-LIB_ReferenceResults/pull/20)

For Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_Braking i'll update the references after the back-port of [#4563](https://github.com/modelica/ModelicaStandardLibrary/pull/4563) is finished.

Esther-Devakirubai, 2025-03-10:
> [@Esther-Devakirubai](https://github.com/Esther-Devakirubai) please also back-port [#4563](https://github.com/modelica/ModelicaStandardLibrary/pull/4563) when merged on master. Then we can close this issue.

Backported by https://github.com/modelica/ModelicaStandardLibrary/pull/4582

casella, 2025-03-10:
Back-port done. 

@MatthiasBSchaefer please re-generate reference results for both `Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SynchronousMachines.SMPM_Braking` and `Modelica.Electrical.Machines.Examples.SynchronousMachines.SMPM_Braking`

casella, 2025-03-11:
All set.

closed 2025-03-11 by casella

AHaumer, 2025-03-28:
> [@AHaumer](https://github.com/AHaumer) it seems pretty clear to me that the final state of this simulation is a singular state where all voltages and currents on the variable resistor get to zero, [...]
@casella your reamrk reminds me;
To avoid the incidence of "singular points" it could help to use a forward threshold voltage Vknee (for Si approx. 0.7 V) to shift the diodes "singular point".
BTW, I do not really understand why that's a "singular point" ...

casella, 2025-04-11:
> BTW, I do not really understand why that's a "singular point" ...

That's a good question. My gut feeling is that you are ending up with a nonlinear system of equations that is badly conditioned because of the very low resistance on one side and conductance on the other side. 

Obviously, if you put Ron = 0 and Goff = 0, the problem _will_ likely be singular (akin to division by zero). As those parameters become smaller and smaller, I guess the condition number of the Jacobian of the nonlinear equation system will become larger and larger, making it more and more numerically challenging to solve it reliably.

We should try develop this kind of arguments in #4577. I believe this is an interesting research topic, which could lead to interesting methodological results with a practical impact on cross-tool portability of Modelica models.

PR #4562 (2025-03-05): Attempt to fix #4559 by increasing RonDiode and GoffDiode
This attempts to fix #4559 by making the system less brutally nonlinear when approaching zero voltage and current on the load towards the end of the transient.