modelica/ModelicaStandardLibrary #4576: Problems with gconstructor.h and source FMU's in MSVC

KarlWernersson, 2025-03-07:
Hi

When trying to compile a Dymola source FMU together with a Dymola simulation and then link them there are many identical functions that causes link problems. A way to simply solve this is by making sure things are static declared. 

Now for the gconstructor for MSVC the constructor has the following hack to force some function calls before main

`#define G_DEFINE_CONSTRUCTOR(_func) G_MSVC_CTOR (_func, G_MSVC_SYMBOL_PREFIX)`

```
#define G_MSVC_CTOR(_func,_sym_prefix) \
  static void _func(void); \
  extern int (* _array ## _func)(void); \
  int _func ## _wrapper(void) { _func(); return _array ## _func == NULL; } \
  __pragma(comment(linker,"/include:" _sym_prefix # _func "_wrapper")) \
  __pragma(section(".CRT$XCU",read)) \
  __declspec(allocate(".CRT$XCU")) int (* _array ## _func)(void) = _func ## _wrapper;
```

As some of these functions aren't static declared i get to problems when both the FMU and the dymola simuation uses modelicaInternal.c

I tried to modify the hack to make it static but it don't seem to work and i cant say I am an expert in this low level mscv hacking.

My workaround was to copy the name pasting of ModelIdentifier from FMI as this for us is only an issue for FMU's for us.  

added to gconstructor _MSC_VER >=1500 defined

```
#ifndef MODEL_IDENTIFIER
#define MODEL_IDENTIFIER
#endif
```

```
#define G_Paste(a,b)     a ## b
#define G_PasteB(a,b)    G_Paste(a,b)
#define G_FullName(name) G_PasteB(MODEL_IDENTIFIER, name)

```
non _MSC_VER

`#define G_FullName(name) name`

and then later in ModelicaRandom,c ModelicaInternal.c, modelicaStandardTables.c I modified the use of  G_DEFINE_CONSTRUCTOR by inserting the G_FullName macro

```
G_DEFINE_CONSTRUCTOR(G_FullName(ModelicaInternal_initializeCS))

static void G_FullName(ModelicaInternal_initializeCS)(void) {
    InitializeCriticalSection(&cs);
}
```

```
G_DEFINE_DESTRUCTOR(G_FullName(ModelicaInternal_deleteCS))
static void G_FullName(ModelicaInternal_deleteCS)(void) {
    DeleteCriticalSection(&cs);
}
```

If anyone has a better solution i am al ears, otherwise i am happy to make a pull request

beutlich, 2025-03-09:
@KarlWernersson I understand that it night be a problem if a source code FMU contains the same source files as the simulation environment. I want to create a minimal example, indpendent of FMU or Dymola. Is it as simple as building an executable which compiles ModelicaRandom.c and links ModelicaExternalc.lib where

1. ModelicaRandom.c mimics the file of the source code FMU
2. Linking ModelicaExternalc.lib (which already contains the symbols of ModelicaRandom) mimics the Dymola simulation environment

beutlich, 2025-03-10:
@KarlWernersson Have a look at my branch [beutlich:issue4576](https://github.com/modelica/ModelicaStandardLibrary/compare/master...beutlich:ModelicaStandardLibrary:issue4576) where I first added some test case to demonstrating the MSVC build error on duplicated symbols and afterwards applied the name mangling following your proposal. Let me know if it works for you if you export FMU and re-import FMU fom this branch.

HansOlsson, 2025-03-11:
@KarlWernersson will look at that next week.

However, I realized that we have to be even more careful regarding this, as there are two interrelated problems:
- Upgrading uses of gconstructor to avoid link-conflicts
- Versioning of gconstructor

The second issue ( related to https://github.com/modelica-3rdparty/ExternData/issues/53 ) is that if you have a library (like ExternData) using the existing `gconstructor.h` having a new file with the same name but different contents will cause problems (especially as it impacts code using it and thus is an "API-change"). Since we cannot immediately change all existing copies of such libraries (including ExternData) I believe that the safest approach is to rename the file when modifying it.

(Obviously those libraries should be upgraded as well at some time.)

beutlich, 2025-03-11:
Yes, it might be better to have a new file and new macros to avoid any potential compatibility problems. I updated my brach accordingly.

GarronFish, 2025-03-14:
Will this mean that older versions of ExtrernData will not be compatible with Dymola 2025x R1 in which these changes have been made?

KarlWernersson, 2025-03-14:
Hi @GarronFish we will try to avoid that.

beutlich, 2025-03-18:
> @KarlWernersson will look at that next week.

Did you find the time to inspect my proposed and test changes? I also wonder if this change is relevant for upcoming MSL v4.1.0?

KarlWernersson, 2025-03-19:
@beutlich 
Thanks for taking you time looking at this and your quick response, i will look at it today or tomorrow, I got struck with a strong case of the flu and was on a business trip so i didn't have time to look at it until now.

beutlich, 2025-04-01:
> Did you find the time to inspect my proposed and test changes? I also wonder if this change is relevant for upcoming MSL v4.1.0?

@KarlWernersson Any news on this?

casella, 2025-04-11:
@beutlich how do we proceed? Shall something be merged on master?