modelica/ModelicaStandardLibrary #4587: Last-minute issues with MSL 4.1.0-beta.2

casella, 2025-03-11:
This ticket is meant to collect potential last-minute issues with MSL 4.1.0-beta.2, before tagging the RC.1 version (scheduled date: Mar 25 2025).

If you have any issue to report please post a comment here, possibly linking it to another issue that has all the details.

beutlich, 2025-03-11:
Edited above.

AHaumer, 2025-03-11:
June 2020 (release 4.0.0) the list of libraries was:

[image]

AHaumer, 2025-03-11:
> Edited above.

Nice to hear. Please explain: How should I test? 
That's a mess. When will this endless problems end?

AHaumer, 2025-03-12:
@casella how shall we proceed, do you have an idea how to overcome the troubles with the tables?
I can't translate a single model that's using a table.

beutlich, 2025-03-12:
> I can't translate a single model that's using a table using 4.1.0 but they work perfectly with 4.0.0.

Can you create a proper ticket with the actual error, tool and version.

AHaumer, 2025-03-12:
> > I can't translate a single model that's using a table using 4.1.0 but they work perfectly with 4.0.0.
> 
> Can you create a proper ticket with the actual error, tool and version.

@beutlich see #4592

maltelenz, 2025-03-17:
I reported https://github.com/modelica/ModelicaStandardLibrary/issues/4596.

GallLeo, 2025-03-18:
I checked the box for PlanarMechanics, because success was reported in #4595

maltelenz, 2025-03-18:
@casella I also want to bring up https://github.com/modelica/MAP-LIB_ReferenceResults/pull/21. I'm guessing that should be merged?

GallLeo, 2025-03-20:
I ticked the box for Clara and ClaraPlus, as reported here: https://github.com/modelica/ModelicaStandardLibrary/issues/4595#issuecomment-2736339842

christiankral, 2025-03-25:
I checked the boxes of `PhotoVoltaics_TGM` and `HanserModelica` (OpenModelica).

casella, 2025-04-01:
> [@casella](https://github.com/casella) I also want to bring up [modelica/MAP-LIB_ReferenceResults#21](https://github.com/modelica/MAP-LIB_ReferenceResults/pull/21). I'm guessing that should be merged?

Yes, sure, just did it. Sorry for the delay.

casella, 2025-04-01:
> I'd consider regression issue [#4598](https://github.com/modelica/ModelicaStandardLibrary/issues/4598) as a blocker.

Definitely, the medium description says that it is valid up to 455 K, but after #3704 the model gives bogus results above the critical temperature. We'll discuss it during the monthly meeting today.

casella, 2025-04-11:
Just ticked the box for MatIO

casella, 2025-04-14:
Everything mentioned in this ticket was fixed or taken care of (in particular updating the Release Notes regarding the libraries tested with MSL 4.1.0), except for #4598, but that has a ticket on its own. We can close this one.

closed 2025-04-14 by casella