modelica/ModelicaStandardLibrary #4592: Compilation fails with tables from 4.1.0

AHaumer, 2025-03-12:
Windows 11 24H2
Dymola 2025x
MSL 4.1.0 beta.2
Compiling and linking the model (Visual C++). 
dsmodel.c
dsmodel.c(206): error C2065: "ModelicaStandardTables_CombiTable1D_init3": nichtdeklarierter Bezeichner
dsmodel.c(206): error C2064: Ausdruck ergibt keine Funktion, die 0 Argumente bernimmt
Error generating Dymosim.

beutlich, 2025-03-12:
It seems you did not set `Advanced.PlaceDymolaSourceFirst = 2;` (to include installed Dymola files after user files) which solves the issue for me.

```
simulateModel(
  "Test.TestCombiTable1Dse",
  numberOfIntervals=0,
  outputInterval=0.0001,
  resultFile="TestCombiTable1Dse");
 = false
Advanced.PlaceDymolaSourceFirst := 2
simulateModel(
  "Test.TestCombiTable1Dse",
  numberOfIntervals=0,
  outputInterval=0.0001,
  resultFile="TestCombiTable1Dse");
 = true
```

AHaumer, 2025-03-12:
Thanks a lot @beutlich !
It seems that I wasn't aware of this switch, since I'm not an experienced programmer like you.
Where did you spread this knowledge?

AHaumer, 2025-03-12:
Since this is marked as a bug and the problem is solved, I'll close the issue.

closed 2025-03-12 by AHaumer

beutlich, 2025-03-12:
If I remember correctly `Advanced.PlaceDymolaSourceFirst := 2` was first mentioned in https://github.com/modelica/ModelicaStandardLibrary/pull/3453#issuecomment-592506380 and since then consequently used in our msl-release scripts, see https://github.com/modelica-tools/msl-release/commit/f230d75a65cf49866062388b23e3c0cada70b547.