modelica/ModelicaStandardLibrary #4595: MLS 4.1.0 Testing with 3rd-party libraries

casella, 2025-03-14:
The Modelica Standard Library 4.1.0-beta.2 was [released](https://github.com/modelica/ModelicaStandardLibrary/releases/tag/v4.1.0-beta.2) on 11/03/2025 and is now available for testing.

MSL version 4.1.0 is backwards compatible with version 4.0.0, which means if your library(es) use MSL 4.0.0, you can switch to MLS 4.1.0 without the need of touching your code or running conversion scripts. A few minor exceptions to backward compatibility are documented in the release notes, but they are very unlikely to be relevant for end users.

If you are a developer of a library that uses MLS 4.0.0, we warmly invite you to check if it works with MSL 4.1.0 and report your experience by posting a comment on this ticket. Please check the GitHub [release page](https://github.com/modelica/ModelicaStandardLibrary/releases/tag/v4.1.0-beta.2) for instructions on how to do that with Dymola and OpenModelica. The draft release notes and list of fixed issues are available as PDF files can be downloaded from the same page.

Thank you!

tobolar, 2025-03-17:
[PlanarMechanics 1.6.0](https://github.com/dzimmer/PlanarMechanics/releases/tag/v1.6.0): run all examples with
- Dymola 2025x, 2024-10-11
- OpenModelica v1.24.2 (64-bit)

and compared to https://github.com/DLR-SR/PlanarMechanics_ReferenceResults.git. Seems be all fine!

NBrandtTLK, 2025-03-19:
I ran regression tests for TIL/TILMedia with MSL 4.0.0 and 4.1.0 in Dymola 2024x Refresh 1 and Dymola 2025x.
All testers translated and simulated successfully. Some testers have more or less state events due to changes in MSL models used, but nothing critical appeared. 

There is an issue with the specific enthalpy of R134a in MSL Media that I opened #4598

NBrandtTLK, 2025-03-19:
We have also tested Clara/ClaraPlus. It works without any problems.

mwetter, 2025-03-19:
I tested the Buildings library 12.0.0. All models work as-is, and the result verification does not flag any differences.

thom-marx, 2025-03-20:
We (XRG Simulation) tested the next release of the Human Comfort Library 2.19.0 in Dymola 2025x without any problems.

HansOlsson, 2025-03-24:
We (Dassault Systemes) tested some of our partner libraries.

The following seemed ok (there were minor regressions except as noted):

-  FlexibleBodies 2.4.0.
- VehicleInterfaces 2.0.1.
- Modelica_StateGraph2 2.1.0
- Modelica_DeviceDrivers 2.1 (without any issues)
- FlightDynamics 1.0.4 (without any issues)
- HumanComfortLib 2.18.0 (next version also reported above by XRG)
- CATIAMultiBody 2.0
- Battery 2.8.0
- Engines 2024.2
- ElectrifiedPowertrains 1.10.0
- Claytex 2024.2
- Cooling 1.5.3 (without any issues)
- VeSyMA 2024.2
- VeSyMA2EPTL 2024.2
- ThermalManagementDemos 1.2.1
- ClaRa\* 1.7.5 (also reported above by TLK)
- FluidDynamicsLib 2.18.0 (next version also reported above by XRG)
- FluidPower 2024.2
- ElectricPowerSystems 1.6.5
- Hydrogen 1.4.1
- VehicleDemos 2024.2
- ClaytexFluid 2024.2 (without any issues)
- PneumaticSystems 1.7.1
- Visa2\* 1.17
- ProcessModeling 1.2.0 (without any issues)
- TIL\* 2024.1 (also reported above by TLK)

The following will need additional analysis (not necessarily problems, but more analysis needed):

* HVACLib 3.3.0 (for 32-bit libraries)
* AviationSystems 1.6.1

One minor issue was noted for the release notes, will add a separate issue for that.

The following libraries have to be updated, but so far it has not revealed any real issue in MSL 4.1.0:

* Suspensions 2024.2
* PTDynamics 2024.2

thom-marx, 2025-03-24:
We found a problem with SmartInt 0.5.1 and SmartInt Plus Library. I created a ticket here: #4602 

It would be great if somebody has the time to check that. Thank you!

StephanZiegler, 2025-03-25:
Some more insights on the Dassault Systemes Libraries.
We successfully tested the following in house libraries, which will be released with the upcoming Dymola 2025x Refresh 1 release:
 
- AviationSystems 1.6.2
- AviationSystems_Demos 1.6.2
- Battery 2.8.1
- BrushlessDCDrives 1.4.3
- Cooling 1.5.4
- DassaultSystemes 1.14.0
- DymolaModels 1.10.0
- ElectricPowerSystems 1.7.0
- ElectrifiedPowertrains 1.11.0
- FTire 1.3.2
- Hydrogen 1.4.2
- ThermalManagementDemos 1.2.2
- ThermodynamicsConnector 1.3.0
- PneumaticSystems 1.7.2
- SDF 0.4.5
- TestingLib 1.10.0
- WindPower 1.1.5

For some we encountered some minor regressions (e.g. due to the modified Modelica.Constants.eps ), but nothing that would require a regression investigation for MSL.

Will the list in the Release Notes be updated and extended accordingly?

casella, 2025-04-11:
> Will the list in the Release Notes be updated and extended accordingly?

Yes, see d84fbe7 and 87a5a4080

closed 2025-04-11 by casella