modelica/ModelicaStandardLibrary #4596: Correctness issue in TestSimpleTank

maltelenz, 2025-03-17:
We have a correctness issue in `ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank` in 4.1.0 when simulating with Wolfram System Modeler.

[image]

[image]

I see that OpenModelica also has a failure in `tank.level`: https://libraries.openmodelica.org/branches/master/ModelicaTest_4.1.0-beta.om/files/ModelicaTest_4.1.0-beta.om_ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank.diff.tank.level.html

@casella Is this expected from the recent changes in the tank models, like [this](https://github.com/modelica/ModelicaStandardLibrary/pull/4527/files#diff-df7deb2fd7bbe61a45433526306790d75eb4bc80b6e542b96166f2124c2b4f7b)? Do we need new reference results?

casella, 2025-04-01:
I guess for some reason we forgot to update the reference results. @MatthiasBSchaefer could you please take care of that? Thanks!

casella, 2025-04-01:
@GallLeo will take care of that.

casella, 2025-04-01:
@maltelenz, @casella, @HansOlsson, please check that you have no regressions on your tools after the results are updated, before closing this ticket.

casella, 2025-04-11:
New references files are now merged in. @maltelenz, can you please double-check and close this issue if all is fine now? Thanks!

closed 2025-04-14 by maltelenz