modelica/ModelicaStandardLibrary #4598: Incorrect specific enthalpy of R134a at high temperatures with MSL 4.1.0

NBrandtTLK, 2025-03-19:
In our library TIL we have an adapter between TIL VLEFluid Media and MSL Media.
The specific enthalpy is calculated for both at 1 bar and 400 K in order to take into account a possible zero point shift of the selected media.
For the tester with 134a our regression test between 4.0.0 and 4.1.0 failed.

parameter Real specificEnthalpy_MSL = Modelica.Media.R134a.R134a_ph.specificEnthalpy(Modelica.Media.R134a.R134a_ph.setState_pTX(1e5,400));

With MSL 4.0.0 this results in a value of 519298 J/kg
With MSL 4.1.0 this results in a value of 2.8e36 J/kg

It seems like the temperature is limited to some Kelvin below critical temperature.

beutlich, 2025-03-19:
Thanks for reporting. Regression was introduced by f1574e8116fd5e4d72c5cbae4176f29e1dc3445f in #3704 to resolve #3695 about four years ago. That issue #3695 needs to be revisited.

I added a new test model in #4599.

casella, 2025-04-01:
Discussion in the MAP-Lib meeting Apr 1 2025:
- #3704 introduces non-backwards compatible changes (breaking the high-temperature computations), while trying to fix another issue. As such it should not get into supposedly backwards-compatible 4.1.0
- we should then revert #3704 and ask XRG if they can provide a last minute fix, otherwise postpone that to either 4.1.1 or 4.2.0

casella, 2025-04-01:
See also the [comment in #3695](https://github.com/modelica/ModelicaStandardLibrary/issues/3695#issuecomment-2769046489)

casella, 2025-04-15:
I made some more analysis with this test model, which scans the pT-plane with different values of pressure and temperature, staying clear from the saturation dome, where setState_pTX cannot be used.

```modelica
model TestR134a_pTX
  package Medium = Modelica.Media.R134a.R134a_ph;
  parameter Medium.AbsolutePressure pmin = 0.004e5;
  parameter Medium.AbsolutePressure pmax = 700e5;
  parameter Modelica.Units.SI.PerUnit p_increase = 2;
  parameter Medium.Temperature Tmin = 240;
  parameter Medium.Temperature Tmax = 450;

  Medium.ThermodynamicState state = Medium.setState_pTX(p,T);
  Medium.Temperature Tsat = Medium.saturationTemperature(p);
  Medium.AbsolutePressure p;
  Medium.Temperature T;
  Medium.Density d = Medium.density(state);
  Medium.SpecificEnthalpy h = Medium.specificEnthalpy(state);
  Medium.ThermalConductivity k = Medium.thermalConductivity(state);
  
  Modelica.Units.SI.PerUnit dT;
initial equation
  p = pmin;
  T = Tmin;
  dT = 100*35;
equation
  der(T) = dT;
  when abs(T - Tsat) < 2 then
    reinit(T, Tsat+4*sign(dT));
  end when;
  when {T > Tmax, T < Tmin} then
    dT = -pre(dT);
    p = pre(p)*p_increase;
  end when;
  when p > pmax then
    terminate("Upper pressure bound exceeded");
  end when;
annotation(
    experiment(StartTime = 0, StopTime = 1, Tolerance = 1e-06, Interval = 0.0001));
end TestR134a_pTX;
```
The temperature goes continuously back and forth between the two limits, whereas the pressure is increased by factors 2X after each temperature scan, until it reaches about 260 bars. 

With Modelica 4.0.0, I get this plot of the specific enthalpy

[image]

clearly, there are problems for certain combinations of pT, which were the reasons behind #3695, that was reporting bogus values in the range 273 to 297 K and 10 to 20 bar. Unfortunately, in the attempt of fixing those problems (which was successful, per se), after #3704 the medium model as a whole became much less reliable: the plot of specific enthalpy with MSL 4.1.0 becomes:

[image]

#3704 indeed fixed the problems that were present in the `setState_pTX` function _in the very narrow pT range that was tested in_  `ModelicaTest.Media.TestOnly.R134a_setState_pTX` (273 to 298 K and 10 to 20 bar). Unfortunately, #3704 screwed up setState_pTX in a much larger portion of the pT plane, basically the entire `T > Tcrit` region

To me it's clear that neither of the two cases is satisfactory, but also that the situation got much worse after #3704. Fixing this problem requires a substantial amount of work and is not feasible for 4.1.0, unless we significantly delay the release, which doesn't seem a good idea to me.

Hence, I stand by my proposal to revert #3704 for MSL 4.1.0, which also resolved the regression reported by TLK. We should of course try to fix #3695 properly with help from XRG, but that will go into 4.1.1 or 4.2.0.

beutlich, 2025-04-15:
If this issue did not get fixed within the last four years I doubt it will be fixed in the future at all. Should the questionable function rather be removed?

casella, 2025-04-15:
It is fixed now 😃

casella, 2025-04-15:
@NBrandtTLK thanks for reporting, this was a fairly serious regression. It should be fine in the final release.

NBrandtTLK, 2025-04-16:
Thanks for fixing the issue. I'm glad I was able to help with reporting.

closed 2025-04-16 by NBrandtTLK