modelica/ModelicaStandardLibrary #4600: MSL 4.1.0 How to count classes for docu?

GallLeo, 2025-03-20:
As discussed in the last MAP-LIB meeting:
I tried to update the class statistics in top-level package.mo:

> This version of the Modelica Standard Library consists of
> - 1417 component models and blocks,
> - 512 example models, and
> - 1219 functions

https://github.com/modelica/ModelicaStandardLibrary/blob/cf4c2e33b950125426e7b910516f512b433170ba/Modelica/package.mo#L68

But, I did not get consistent figures.
Question marks indicate numbers I do not understand:

```
       component models & blocks, example models, functions
4.1.0  1466?                      534             1354? 1219?
4.0.0  1417                       512             1219?
3.2.3  1288                       404             1227  
```

Source 1: Counts in Dymola 
via `ModelManagement.Structure.AST.Examples.countModelsInPackage("Modelica");`

```
       {models, blocks, function, type, package}
v4.1.0 {1602,   398,    1354,     678,  756}
v4.0.0 {1572,   392,    1354,     666,  748}

```

Source 2: Regression test https://msl.ltx.de
```
                                  example models
4.1.0                             534
4.0.0                             534 (already includes new examples from MSL 4.1.0, because they would not have any reference, otherwise)
```

Calculations:
- component models & blocks = (models+blocks)-examples 
  - v4.1.0: 1602+398-534 = 1466
  - Check with v4.0.0: 1572+392-512=1452 -> does not match, +14 blocks seems reasonable for 4.0.0->4.1.0

@beutlich Do you remember, how the last figures have been created?
@HansOlsson Do you have an idea, why Dymola counts so many more functions than documented so far? Partial functions & functions in models seem to be ignored, correctly.

Probably we should add a function to https://github.com/modelica-tools/msl-release ?

beutlich, 2025-03-20:
> Probably we should add a function to https://github.com/modelica-tools/msl-release ?

What's wrong with the existing function `ReleaseChecks.countClassesInPackage`?

https://github.com/modelica-tools/msl-release/blob/c45448265222c73da9f819b286c4d9ce6797941f/ReleaseChecks.mo#L75

GallLeo, 2025-03-31:
Thomas' `ReleaseChecks.countClassesInPackage();` works fine :)

```
       {models, blocks, functions, types, packages, examples}
v4.1.0 {1036,   394,    1219,      676,   719,      532}
v4.0.0 {1029,   388,    1219,      664,   715,      512}
```

Calculation:
component models & blocks = (models+blocks)
- v4.1.0: 1036 + 394 = 1430
- v4.0.0: 1029 + 388 = 1417

Figures for `package.mo`
```
       component models & blocks, example models, functions
4.1.0  1430                       534             1219
4.0.0  1417                       512             1219
3.2.3  1288                       404             1227  
```

For release notes:
- 13 new component models & blocks
- 22 new example models

GallLeo, 2025-03-31:
Todo:
- [ ] Merge #4605 
- [ ] Add to release wiki
- [ ] Backport

GallLeo, 2025-03-31:
Release notes
https://github.com/modelica/ModelicaStandardLibrary/releases/download/v4.1.0-beta.2/ReleaseNotes_MSL_4.1.0-beta.2.pdf
list 17 new components.

But, we count 13.

Did we delete components (which should not be the case in a minor release)?

beutlich, 2025-04-01:
Three models Modelica.Electrical.Analog.Lines.{TLine1, TLine2, TLine3} have been set to obsolete and thus are no longer counted since only public, non-partial, non-internal and non-obsolete classes are consiered. So yes, it's 16 new models/blocks and one new type.

Maybe the three obsoletes models should be added to the release notes as well if not yet already done.

casella, 2025-04-11:
#4605 merged and back-ported by #4615

closed 2025-04-13 by beutlich

beutlich, 2025-05-20:
@nithish-selvan FYI