modelica/ModelicaStandardLibrary #4602: Compilation problems with SmartInt 0.5.1 and SmartInt Plus Library in Modelica 4.1.0

thom-marx, 2025-03-24:
A few models in the SmartIntLibrary 0.5.1 cannot be compiled in Dymola 2025x with Modelica 4.1.0:

e.g.

SMArtInt.Tester.ExamplePI.ReferenceModels.DoubleRoom_Discrete
SMArtInt.Tester.ExamplePI.ReferenceModels.DoubleRoom_ContinuousPI

The models consist of MSL Components only.

The error message is the following:

Compiler message:

Compiling and linking the model (Visual C++). 

dsmodel.c
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_minimumTime already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_maximumTime already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_minimumAbscissa already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_maximumAbscissa already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_minimumAbscissa already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_maximumAbscissa already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_close already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_close already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_close already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_getValue already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_getDerValue already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_getDer2Value already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_nextTimeEvent already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_read already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_getValue already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_getDerValue already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_getDer2Value already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_read already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_getValue already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_getDerValue already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_getDer2Value already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_read already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_init2 already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_init2 already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_init2 already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTimeTable_init already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable1D_init already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
ModelicaStandardTables.lib(ModelicaStandardTables.obj) : error LNK2005: _ModelicaStandardTables_CombiTable2D_init already defined in ModelicaExternalC.lib(ModelicaStandardTables.obj)
Creating library dymosim.lib and object dymosim.exp
dymosim.ex_ : fatal error LNK1169: one or more multiply defined symbols found

Error generating Dymosim.

The SmartInt library is Open Source and can be downloaded here: 

https://github.com/xrg-simulation/SMArtIInt

I also linked an example model as 

[DoubleRoom_ContinuousPI.txt](https://github.com/user-attachments/files/19426415/DoubleRoom_ContinuousPI.txt)

.txt to this ticket.

Thank you in advance!

HansOlsson, 2025-03-24:
Should be resolved by: https://github.com/modelica/ModelicaStandardLibrary/issues/4293#issuecomment-1918625043

thom-marx, 2025-03-24:
Thanks for your answer. In Dymola 2024x the models work. Maybe it is because of my missing "Fix Pack 2" in Dymola 2025x? Sorry for that.

casella, 2025-04-11:
@beutlich I understand the problem is that you need the right version of Dymola, not an MSL issue.

closed 2025-04-11 by casella