modelica/ModelicaStandardLibrary #4604: Review and improve the CombiTables of FMI-LS-STRUCT from a Modelica side

chrbertsch, 2025-03-26:
Please review https://github.com/modelica/fmi-ls-struct
see also https://github.com/modelica/fmi-ls-struct/issues/18

beutlich, 2025-03-27:
First remarks:

* Instead of referencing the outdated page https://doc.modelica.org/om/Modelica.Blocks.Tables.html (OF MSL v3.2.3) the latest published version for v4.0.0 is found at https://doc.modelica.org/Modelica%204.0.0/Resources/helpOM/Modelica.Blocks.Tables.html.
* Note that it might be useful to hide this link behind a permalink.

HansOlsson, 2025-04-22:
Please explain why anyone from Modelica side should bother when the review and https://github.com/modelica/fmi-ls-struct/issues/18 is ignored without any feedback?

beutlich, 2025-04-22:
More remarks:

- Layout issue for `CUBE_5` in https://modelica.github.io/fmi-ls-struct/main/#_definitions_2
  
  [image]

- First listing of https://modelica.github.io/fmi-ls-struct/main/#_example: Should `in_y` be of causality=local and given a start attribute?

  [image]

beutlich, 2025-04-22:
Remarks on naming:

* I would actually drop the MSl reference as origin of the "org.fmi-standard.fmi-ls-struct.map.combitable1d" or "org.fmi-standard.fmi-ls-struct.map.combitable2d".
* AFAIK the term _combi_ does not mean that coordinate vectors and matrix data are combined to one amalgamated matrix. Instead the term _combi_ originates from the the table origin handling, either from file I/O or from model.
* I asked LLM about better names: https://chatgpt.com/share/6807c052-9758-8005-ba9a-fac56fbd3728
  [image]
* In that case we only need one specifier, say org.fmi-standard.fmi-ls-struct.map.tabularGrid".
* MSL 1-d and 2-d lookup tables can now served solely as examples.