modelica/ModelicaStandardLibrary #4608: Warning in Electrical.Polyphase.Functions

AHaumer, 2025-04-01:
The functions Modelica.Electrical.Polyphase.Functions.{factorY2D, factor Y2DC} use an intermediate (protected) variable `mBasic` calculated from the funtion's input `m`. This variable `mBasic` has been declared as `parameter`.
Dymola now complains "The function parameter mBasic has a binding equation which is not a parameter expression.", but this didn't happen in previous versions and it doesn't affect the results. OpenModelica has no complaints. 
For sure this is not a critical issue, but should I fix that for the next release by removing `parameter` from the variable's defintion?
@HansOlsson @casella waht's your opinion?

beutlich, 2025-04-01:
Duplicate of #4402.

closed 2025-04-01 by beutlich

AHaumer, 2025-04-01:
Thanks to @maltelenz I didn't rememebr that!