modelica/ModelicaStandardLibrary #4637: Comparabilty of chaotic behaviour

AHaumer, 2025-04-21:
During regression testing we had a short discussion about comparabilty of chaotic behaviour - if I remember correctly, Modelica.Electrical.Analog.Examples.ChuaCircuit. Here we decided to shorten the interval for comparison to exclude chaotic behaviour. In my opinion it is possible in many cases to name parameterization for periodic behaviour which can be compared very well and different parameterization to demonstrate chaotic behaviour. I have implemented a small library [ChaoticCircuits](https://github.com/AHaumer/ChaoticCircuits) to demonstrate Chuas ciruit and three other examples from literature. They run fine in OM and Dymola. I'll try to provide reference results, too. Maybe @maltelenz could you compare using WSM?
@casella what's your opinion after having a look at the library? 
I would suggest to include these models into Modelica.Electrical.Analog.Examles.

maltelenz, 2025-04-22:
@AHaumer WSM gets different results for the models in `ChaoticCircuits.ColpittsCircuit`. Here is a plot of `iL` at the beginning of `AnalyticEquations`:

[image]

Or at the end:

[image]

Similar differences are present in the other variables, and in the `IdealCircuit` and `ImprovedCircuit`.

For `ChaoticDiode`, `ChuasCircuit` and `ChaoticOscillator`, the results from WSM match the reference results.

AHaumer, 2025-04-24:
See also discussion in #4577

HansOlsson, 2025-04-25:
I would prefer to keep the models chaotic (if they are intended for that), and adapt the comparisons to only run them a short while.

Obviously one can imagine more advanced solutions - but to me:
- The examples are intended as examples, thus an example of a chaotic system should show chaotic behavior.
- The comparisons are intended to ensure that models don't misbehave too much; we cannot expect bit-equality in simulation results; so only comparing the first part will be good enough to detect that.

Obviously we can do more, but trying to find more stable solution will be a fine balancing act (and it's not clear that periodic solutions are inherently more stable than chaotic ones) and I think the effort should be spent elsewhere. Comparing the stable attractors themselves is also possible, but that would require a completely different comparison tool - and I don't see the benefit for MSL-development in setting up that.

AHaumer, 2025-04-25:
@HansOlsson there are oscillators designed for stable periodic behaviour. 
However, some scientists investigate them how to obtain chaotic behaviour to exploit chaos in various sciences, e.g. cryptograhpy.

AHaumer, 2025-04-25:
I adapted a little bit the original Modelica.Electrical.Analog.Examples.ChuaCircuit, comparing the results with Dymola2025x and OpenModelica v1.25.0:
For periodic behaviour the maximum relative deviation is about 0,5%. Is this good enough for comparability?
Changing one parameter (R=1500 Ohm instead of R=1900 Ohm) reveals chaotic behaviour. Yes the results run apart after short time.
We could even reach a steady-state solution with constant states using R=2000 Ohm. 
Here are some small differences during initial transients but both reach the same terminal states.

I have attached the model and a reference result obtained for period behaviour with Tolerance=1e.7,
@maltelenz may I kindly ask you to have a look how close WSM's results are to the reference?

[ChuaCircuit.zip](https://github.com/user-attachments/files/19908085/ChuaCircuit.zip)

HansOlsson, 2025-04-25:
> I adapted a little bit the original Modelica.Electrical.Analog.Examples.ChuaCircuit, comparing the results with Dymola2025x and OpenModelica v1.25.0: For periodic behaviour the maximum relative [...]

Ah, so different parameters - not just different starting point.

As indicated I don't think that makes sense as example, as we would have a big explanation saying "this is normally seen as a chaotic circuit; but it is not chaotic for the given parameters, to get chaos you have to set these parameters...".
That would be a step backwards - an example should demonstrate things as default, not requiring modifications.

Basically: I view examples as examples that as default should demonstrate something when you just simulate it, not primarily as testing models.

HansOlsson, 2025-04-25:
BTW: After simulating Colpitt oscillator I feel confident to say that it is not chaotic for those parameters, it's just a transient (and depending on tolerance, numerical noise) that keeps if from a perfect periodic solution.

AHaumer, 2025-04-25:
For me a compromise looks fine: Default parameters for periodic behaviour, special parameters for chaos. 
Anyhow it's interesting to study the travel into chaos. 
Regarding Colpitt:
M.P. Kennedy reported 1994 [Chaos in the Colpitts oscillator](https://ieeexplore.ieee.org/document/331536)
Buscarino describes it in his [Concise Guide to Chaotic Electronic Circuits](https://link.springer.com/book/10.1007/978-3-319-05900-6)

AHaumer, 2025-04-26:
I compared the analytic equations of Chua's Circuit and Colpitt's Oscillator, plotting the phase diagram v2 vs. v1:
Chua with R=1900 Ohm and R=1500 Ohm:
[image] [image]
Colpitt with RE={50, 100, 200, 300, 500, 2000, 5000} Ohm:
[image] [image] [image] [image] [image] [image] [image]
Looks like bifurcation.

maltelenz, 2025-04-28:
> I adapted a little bit the original Modelica.Electrical.Analog.Examples.ChuaCircuit, comparing the results with Dymola2025x and OpenModelica v1.25.0: For periodic behaviour the maximum relative [...]

WSM results are extremely close to the reference. I didn't check exhaustively, but some manual sampling towards the end were within 0.3%.

AHaumer, 2025-04-28:
Thanks a lot @maltelenz I appreciate your support!
This proves: At least this model can be compared with parameterization for periodic results, and bring chaotic results with changing one parameter. On the path from periodic to chaotic behaviour one could even see different modes (bifurcation) - this parameter variation has to be done by the user.

henrikt-ma, 2025-04-28:
I must say I agree with @HansOlsson; if a system is famous because of its ability to exhibit chaotic behavior, then our testing setup needs to support having default parameters exhibiting the chaotic behavior.  As suggested in [#3473](https://github.com/modelica/ModelicaSpecification/issues/3473) and many times before, the solution to this problem should be to have something like `annotation(TestCase(StopTime = 0.5))` for setting a duration short enough to make it possible to compare trajectories over time in the usual manner.

Thus, I suggest we close this issue and move the discussion to https://github.com/modelica/ModelicaSpecification/issues/3473.

AHaumer, 2025-05-12:
I have investigated 10 chaotic circuits up to now: [https://github.com/AHaumer/ChaoticCircuits](https://github.com/AHaumer/ChaoticCircuits) including Lotka-Volterra, van der Pol, Lorenz, Rössler, Chua, chaotic diode (Pham2016), chaotic oscillator (Tamasevicius2005), Colpitt, Shinriki, Jerk system. 
Some of them were designed to deliver chaotic trajectories. some other as harmonic osciallators that can for certain parameter combinations reveal chaotic behaviour and some other are just equations for pyhsical effects that can be translated to electronic circuits and that can again for certain parameter combinations reveal chaotic behaviour.
All of them are parameterized to deliver periodic results after an initial transient, and they seem to be comparable between simulators. Changing a parameter gradually, you can watch the path to chaos.
I'm pretty sure some of them won't offer a comparable initial transient when parameterized for chaotic results.

henrikt-ma, 2025-05-12:
> I'm pretty sure some of them won't offer a comparable initial transient when parameterized for chaotic results.

At least to me, this always seemed like a matter of choosing a suitable duration, so it would be very valuable if you could (or concluded that you couldn't) support this with actual examples.

AHaumer, 2025-05-13:
@henrikt-ma I think you are right, we always can choose a very short time span for comparison.
But for me the question is what is a better comparison:
Either set one parameter to "periodic" and compare a periodic result with initial transient for a longer period of time 
or set one parameter to "chaotic" and compare the initial transient of the chaotic result for a short period of time before chaos starts.

[image]

[image]