modelica/ModelicaStandardLibrary #4667: Use of parameter without value in Spice3.Internal.MOS2

perost, 2025-06-23:
[Modelica.Electrical.Spice3.Internal.MOS2](https://github.com/modelica/ModelicaStandardLibrary/blob/c2dabe7c979b321760bfc99ad8cc99d2a8777624/Modelica/Electrical/Spice3.mo#L4644) contains these parameter declarations:
```Modelica
final parameter Spice3.Internal.Mos2.Mos2ModelLineParams p=
      Spice3.Internal.Mos2.mos2RenameParametersRevised(modelcard) annotation(Evaluate=true);

final parameter Spice3.Internal.Mos2.Mos2ModelLineParams p1=
      Spice3.Internal.Mos2.mos2ModelLineParamsInitEquationsRevised(p, m_type);
```
`p` is initialized by a call to [mos2RenameParametersRevised](https://github.com/modelica/ModelicaStandardLibrary/blob/c2dabe7c979b321760bfc99ad8cc99d2a8777624/Modelica/Electrical/Spice3.mo#L8502), which does not assign a value to `p.m_bulkCapFactorIsGiven` (nor does that parameter have a start value). `p` is then used as argument to [mos2ModelLineParamsInitEquationsRevised](https://github.com/modelica/ModelicaStandardLibrary/blob/c2dabe7c979b321760bfc99ad8cc99d2a8777624/Modelica/Electrical/Spice3.mo#L7881), which at the end of the function has an if-statement using `m_bulkCapFactorIsGiven` in the condition.

Unless I'm missing something this looks like `p.m_bulkCapFactorIsGiven` will always be used uninitialized in this model. This model is only used by `Spice3.Semiconductors.M_NMOS2` and `Spice3.Semiconductors.M_PMOS2`, neither of which seems to be part of any example models (or anything else).

HansOlsson, 2025-06-23:
One possibility is that:
-  `m_bulkCapFactorIsGiven` should have a start-value of 0 similarly as `m_bulkCapFactor`;
- `mos2RenameParametersRevised` should use similar logic for it as preceding declarations (as in `mos1RenameParameters`)

However, `CJ` has a default of zero - whereas `CBS` has a default of -1e40.

kmerbcn, 2025-06-24:
It is a voltage depending formula relating other spice parameters.

kmerbcn, 2025-06-24:
Hello Francesco

Is this the right place to contribute for the standard modelica library?