modelica/ModelicaStandardLibrary #4688: Problem with the test of Modelica/Electrical/Machines/Examples/InductionMachines/IMC_DCBraking

uschna, 2025-08-21:
There is a problem with the test of `Modelica/Electrical/Machines/Examples/InductionMachines/IMC_DCBraking`. If we reduce the tolerances the results will change. According to `Modelica/Electrical/Machines/Examples/InductionMachines/IMC_DCBraking/creation.txt` (in branches v4.0.0 and v4.1.0) the used tolerance is `1e-7`.
With this tolerance the last value of "`imc.idq_sr[1]`" in `IMC_DCBraking.csv` is `-15.240146546765066`.
But with smaller tolerances (for example `1e-10`) this value is approximately `-14.1`.
Could you please compute and save the reference results with `Tolerance=1e-10`?
Thank you.

henrikt-ma, 2025-08-22:
I made a quick experiment with System Modeler.  For a variable with dynamic range of approximately 280, I aimed for an absolute error tolerance of 1e-3 * 280 = 0.28.  With tighter tolerance and when using DASSL for integration, the final result appears to approach −14.125 from below.  When using CVODES for integration I get somewhat different results, but both methods agree on the final value -14.125 at tolerance 1e-10.  At 1e-8 I was above −14.125 - 0.28 with DASSL (and as usual CVODES was consistently performing better), so based on this I would have set `Tolerance = 1e-8` in the `experiment`-annotation, and used 1e-10 for reference result creation.

AHaumer, 2025-09-07:
Let me explain what happens:
The current in the windings and therefore the Clark-transformed space phasor w.r.t. the stator-fixed coordinate system is impressed. 
The transformation to the rotor-fixed coordinate system depends on the mechanical angle imc.phiMechanical. 
The dependency of imc.phiMechanical on Tolerance is rather weak. 
But if the angle is either near n*pi or pi/2+n*pi either sine or cosine change significantly even with small changes of the angle. 
Therefore currents w.r.t. the rotor-fixed coordinate system shouldn't be used for comparison, especially only comparing final values.
I think we should't change the Tolerance (with could be dependent on the algorithm, the tool, ...)
@uschna do you agree?

Thanks to @johhell another problem was detected - no idea why this didn't some uo during the last tests:
This example won't simulate in OpenModelica. 
I have prepared an adapted version - see enclosed. @uschna could you please test?
Then I would prepare a PR. 
BTW we are searching for MAP-Lib members which are willing to review PRs.

AHaumer, 2025-09-08:
@ushna 
Yes but different algorithms react differently on changing the tolerance.
So you might get a satisfying result with one algorithm and a certain tolerance, but with another algorithm you need tighter tolerance. Consodering different tools and different algorithms I'm against changing the tolerance. Especially if this is driven by comparing a final value and if the comparison signal is not appropriate.
Sorry I did'nt upload the zip.

[IMC_DC_Braking.zip](https://github.com/user-attachments/files/22207167/IMC_DC_Braking.zip)

uschna, 2025-09-08:
@AHaumer: Thank you.
> So you might get a satisfying result with one algorithm and a certain tolerance, but with another algorithm you need tighter tolerance. 

This is in general true. But in the specific case all tools have the same problem, that the tolerance are to big. That means, the results are changing for smaller tolerances.

> Especially if this is driven by comparing a final value and if the comparison signal is not appropriate.

I'm agree, that it is also possible to change, which variables are result variables.

uschna, 2025-09-09:
Dear @AHaumer,
do I understand it correctly, that I should replace
Modelica\Electrical\Machines\Utilities\DcBrakeSettings.mo
with
DcBrakeSettings.mo
and
Modelica\Electrical\Machines\Examples\InductionMachines\IMC_DCBraking.mo
with
IMC_DCBraking.mo
from
DCBrak.zip?
Thank you.
[DCBrak.zip](https://github.com/user-attachments/files/22233241/DCBrak.zip)

uschna, 2025-09-09:
Dear @AHaumer,
in this case I tested it in SimulationX with different solvers. In the most cases the calculation and validation was successful and valid.